bblanchon/ArduinoJson #1723: Nested arrays and structures

simonemarin, 2022-03-03:
**Describe the issue**  
I am trying to parse an array inside the json object and setting the data in 2 structures.
JSON model:
```
{
    "cameras": [
      {
        "viscaip": "192.168.5.163",
        "port": 1259,
        "channel": 1,
        "profile": "0"
      },
      {
        
        "viscaip": "192.168.5.163",
        "port": 1259,
        "channel": 10,
        "profile": "0"
      }
    ],
    "hostip": "192.168.5.102",
}
```

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  
<details><summary>ArduinoJson Troubleshooter's report</summary> <ol><li>The issue happens at compile time</li><li>The error is not in the list</li></ol></details>

**Environment**  
Here is the environment that I'm using':
* Microconroller: ESP32
* Core/runtime: ESP32 2.0
* IDE: Arduino IDE 1.8.19

**Reproduction**  
Here is a small snippet that demonstrate the problem.

```c++
struct Camera {
  char viscaip[16];
  int port;
  int channel;
  char HEXpan[8];
  char HEXtiltUp[8];
  char HEXtiltDown[8];  
};
struct Config {
  char hostip[16];
  Camera* cam[];
};
Config config;
  StaticJsonDocument<512> doc;
  DeserializationError error = deserializeJson(doc, file);
  JsonObject camera = doc["cameras"];
  for (JsonObject::iterator it=camera.begin(); it!=camera.end(); ++it) {
    Log.noticeln("1 camera here");
    Serial.println(camera["channel"].as<int>());
    config.cam[it]->channel = camera["channel"].as<int>(); // lost here ;)
  }
```

**Program output**  
Expected output:
(apparently I get the data right but I cannot set the structure correctly

Actual output:

```
1
10
```

simonemarin, 2022-03-04:
So I have added my own counter but isn't there a way to get the index of the current item?
```
  int counter = 0;
  for (JsonObject camera : doc["cameras"].as<JsonArray>()) {
    Log.noticeln("1 camera here");
    Serial.println(camera["channel"].as<int>());
    config.cam[counter].channel = camera["channel"].as<int>();
    counter++;
  }
```

bblanchon, 2022-03-04:
Hi @simonemarin,

No, there is no way to get the index from the nested object.
Adding a counter, as you did, is the right solution.

Did you solve your original problem?

Best regards,
Benoit

simonemarin, 2022-03-04:
I am still failing ad adding the values from json to the structure:
```
struct Camera {
    char viscaip[16];
    int port;
    int channel;
    char HEXpan[8];
    char HEXtiltUp[8];
    char HEXtiltDown[8];  
};

struct Config {
    char hostip[16];
    Camera* cam[];
};

Config config;
Camera cam[10]; //seems like I HAVE to set a maximum size

StaticJsonDocument<512> doc;
DeserializationError error = deserializeJson(doc, file);
JsonObject camera = doc["cameras"];
for (JsonObject::iterator it=camera.begin(); it!=camera.end(); ++it) {
    Log.noticeln("1 camera here");
    Serial.println(camera["channel"].as<int>());
    config.cam[it]->channel = camera["channel"].as<int>(); // lost here ;)
}

readFile(SPIFFS, filename, config);

void readFile(fs::FS &fs, const char * path, Config &config) {

  File file = fs.open(path);
  if (!file) {
    Log.errorln("Failed to open file for reading");
    return;
  }

  StaticJsonDocument<1024> doc;

  // Deserialize the JSON document
  DeserializationError error = deserializeJson(doc, file);
  if (error)
    Log.errorln(F("Failed to read file, using default configuration"));

  
  for (JsonObject camera : doc["cameras"].as<JsonArray>()) {
    Log.noticeln("Json camera with channel: %d", camera["channel"].as<int>());
    config.cam[cam_counter].channel = camera["channel"].as<int>();
    cam_counter++;
  }

  for (int i = 0; i < cam_counter; i++) {
    Log.noticeln("Iteration %d, struct camera channel: %d", i, cam[i].channel);
  }
  
  Log.noticeln("Total cameras: %d", cam_counter);

}
```
I get :
```
I: Json camera with channel: 1
I: Json camera with channel: 10
I: Iteration 0, struct camera channel: 0
I: Iteration 1, struct camera channel: 0
I: Total cameras: 2
```
So something is wrong in the way I set the data.

simonemarin, 2022-03-04:
yay
`config.cam[cam_counter].channel = camera["channel"].as<int>();`
should be
`cam[cam_counter].channel = camera["channel"].as<int>();`
surely your book will come at hand ;)

bblanchon, 2022-03-05:
Hi @simonemarin,

From your last message, I understand that your problem is fixed, so I'm going to close this issue.

For simple configurations files, you can look at the [JsonConfigFile.ino](https://arduinojson.org/v6/example/config/) example.
For complex configurations, look at the "Configuration in SPIFFS" case study in [Mastering ArduinoJson](https://arduinojson.org/book/): it shows how to leverage [custom converters](https://arduinojson.org/news/2021/05/04/version-6-18-0/) to support nested objects.

@simonemarin, thank you very much for purchasing my book; your support is greatly appreciated.
Let me know if you have any remarks about the book.

Best regards,
Benoit

closed 2022-03-05 by bblanchon