bblanchon/ArduinoJson #1724: Rouding of floas to save payload size

mfalkvidd, 2022-03-04:
**Is your feature request related to a problem? Please describe.**  
A float is unable to represent a lot of numbers exactly. This is counter-intuitive for most humans, but a fact of life for floats.
Examples:
|  float  | serialized result  |
|---|---|
| 23.80  |  23.79999924 |
|  23.90 | 23.89999962  |
|  20.30 | 20.29999924  |
| 19.70 | 19.70000076 |
| 19.80 | 19.79999924 |

In my application, I need to minimize how much data is sent over the internet connection. The API I send my data to accepts JSON, so therefore I prefer to send JSON instead of a more space-efficient binary format.

**Describe the solution you'd like**  
I'd like a way to specify that I only want x decimals when serializing a float.

**Describe alternatives you've considered**  
https://arduinojson.org/v6/how-to/configure-the-serialization-of-floats/ suggests to round the numbers, but that method does not work for numbers that can't be represented exact in a float (such as the numbers in the table above).

I tried setting `ARDUINOJSON_USE_DOUBLE` to `1`. 

I could use a more space-efficient format than JSON, for example CBOR but then I would have to create a CBOR to JSON proxy because the API I am sending to only accepts JSON.

**Additional context**  
I understand that this might not be high priority. Maybe just add a note on https://arduinojson.org/v6/how-to/configure-the-serialization-of-floats/ explaining that the method described does not always work (due to how floats are stored).

I'm using an esp32.

bblanchon, 2022-03-05:
Hi @mfalkvidd,

I just tested the values you included in your message, and here is the result:

```json
[
  23.8,
  23.9,
  20.3,
  19.7,
  19.8
]
```

As you can see, each value is rounded as expected.

See this online demo: https://wandbox.org/permlink/yVV3PZW7XXO7Y4Uq

There must be something special with your setup.
Please make sure you're using the latest version of ArduinoJson and provide a repro if the problem persists.

Best regards,
Benoit

bblanchon, 2022-03-05:
PS: I cannot run the code on an ESP32 right now, so I tried on Wokwi:
https://wokwi.com/arduino/projects/325283324439298644
As you can see, the result is still correct.

mfalkvidd, 2022-03-05:
https://wandbox.org/permlink/gx4ifjGSTk51lG0O the behavior only happens when passing a float variable.

mfalkvidd, 2022-03-05:
More complete repro (I am on mobile so editing was a bit limited): https://wandbox.org/permlink/IFVp0e1Mh2lZMSKK

mfalkvidd, 2022-03-05:
Interestingly, setting ARDUINOJSON_USE_DOUBLE to 0 yields short result: https://wandbox.org/permlink/kDpfmIjhVdzh0Yb9
Not defining ARDUINOJSON_USE_DOUBLE at all will result in the long versions again.

bblanchon, 2022-03-06:
[ARDUINOJSON_USE_DOUBLE](https://arduinojson.org/v6/api/config/use_double/) defaults to 1 since [ArduinoJson 6.19.0](https://arduinojson.org/news/2022/01/08/arduinojson-6-19-0/).

In summary, you can easily fix your problem by either:

* using `double` variables instead of `float`, or,
* defining `ARDUINOJSON_USE_DOUBLE` to 0

I don't think there is much to do on the library side, so we can probably close this issue.

mfalkvidd, 2022-03-06:
Agreed.

closed 2022-03-06 by mfalkvidd