bblanchon/ArduinoJson #1727: deserializeJson sometimes modifying the keys by replacing the first character with "?"

jpswensen, 2022-03-10:
**Describe the bug**  
A particular array of data is having the first character of each key corrupted with a question mark. For example, a key "name" becomes "?ame".

**Troubleshooter report**  
<details><summary>ArduinoJson Troubleshooter's report</summary> <ol><li>The issue happens at run time</li><li>The issue concerns deserialization</li><li><code>deserializeJson()</code> returns <code>Ok</code></li><li>The strings in the <code>JsonDocument</code> change for no reason</li><li>Input type is <code>String</code></li></ol></details>

**Environment**  
Here is the environment that I used:
* Microcontroller: ESP32-C3
* Core/runtime: arduino-esp32 2.0.2 (esp-idf 4.4)
* IDE: VSCode/PIO

**Reproduction**  

Here is a small snippet that reproduces the issue.

```c++
String input = \
      "{\"sports\": \
    [               \
        {           \
     \"name\": \"Cross Country (M)\", \
      \"status\": 0 \
    },    \
    {     \
      \"name\": \"Cross Country (W)\",    \
      \"status\": 0   \
    } \
  ] \
}";

  deserializeJson(doc, input);

  JsonArray sports = doc["sports"].as<JsonArray>();
  

  for (JsonObject sport : sports)
      {        
        serializeJsonPretty(sport,Serial);
        //String name = sport[F("name")];
        const char* name = sport["name"];
        int status = sport[F("status")];
        Serial.print(F("Sport: ")); Serial.print(name);
        Serial.print(F("  Status: ")); Serial.print(status);
        Serial.print(F("  Enabled?: ")); Serial.println(isSportEnabled(name));
      }```

**Program output**  
It is interesting that if I run the snippet of code above at the beginning of the program, it parses as expected. However, if I then do the rest of my startup tasks (NTP, webserver setup, connect to WiFi, start 2 FreeRTOS tasks), then subsequent execution of the exact same code gives the erroneous output

Output when run at startup:

```
20:57:45.071 > {
20:57:45.071 >   "name": "Cross Country (M)",
20:57:45.071 >   "status": 0
20:57:45.077 > }Sport: Cross Country (M)  Status: 0  Enabled?: 0
20:57:45.077 > {
20:57:45.082 >   "name": "Cross Country (W)",
20:57:45.082 >   "status": 0
20:57:45.082 > }Sport: Cross Country (W)  Status: 0  Enabled?: 0
```

Output when run later:

```
20:58:29.642 > {
20:58:29.642 >   "?ame": "Cross Country (M)",
20:58:29.642 >   "status": 0
20:58:29.642 > }Sport:   Status: 0  Enabled?: 0
20:58:29.648 > {
20:58:29.648 >   "?ame": "Cross Country (W)",
20:58:29.655 >   "status": 0
20:58:29.655 > }Sport:   Status: 0  Enabled?: 0
```

jpswensen, 2022-03-10:
Update: I can get the error to go away by using a StaticJsonDocument of the appropriate size, declared as a global variable. Not sure why the DynamicJsonDocument was having corrupted data (and a StaticJsonDocument inside the function was too big).

bblanchon, 2022-03-10:
Hi @jpswensen,

This sounds like a use-after-free, probably because you keep references to a destructed `JsonDocument`.
How do you keep the `DynamicJsonDocument` alive?

Also, make sure to read: [Why must I create a separate config object?](https://arduinojson.org/v6/faq/why-must-i-create-a-separate-config-object/)

Best regards,
Benoit

jpswensen, 2022-03-10:
I guess my example above was a bit faulty as it was the toy problem I had been trying to demonstrate with. I had actually been receiving the string from a webserver and then creating the DynamicsJsonDocument based on the string size.

```
size_t size = input.length();         
DynamicJsonDocument doc(3*input);  
```

I guess I just assumed that there was an internal malloc/new in the DynamicJsonDocument constructor and that it got cleaned up after it went out of scope in the destructor. 

I do have a separate class that has a single global instance where I extract all the pertinent values from the JsonDocument as basic types (float, int, String) and don't store them as JsonObject. 

So, I am still a little confused as to why:

1. This works correctly with the DynamicsJsonObject the first time it is called before any other of my program setup is done.
2. This doesn't work with the DynamicsJsonObject anytime after I have the rest of my program setup completed.
3. This works for a globally declared StaticJsonDocument 

I definitely have a workaround for now using the global StaticJsonDocument, and since the function that is querying the server and parsing the json is threadsafe/can't be called multiple times, I don't have to worry about clobbering that global instance of StatisJsonDocument.

I am wondering if there is some other library I am using that is somehow writing beyond its bounds.

As a bizarre sidenote, I was messing around with the contents of the json input example above and found that if I make the following change

```
String input = \
      "{\"sports\": \
    [               \
        {           \
     \"Xname\": \"Cross Country (M)\", \
      \"status\": 0 \
    },    \
    {     \
      \"name\": \"Cross Country (W)\",    \
      \"status\": 0   \
    } \
  ] \
}";
```

then the key for the first item comes back as "?name" and the keys for all the subsequent items in the json array have the correct key of "name" (which I also tried with a longer example sequence).

Any other thoughts on how to debug this? I drilled down into the source code in JsonDeserializer.hpp and put some print statements in parseKey, parseStringValue, and parseQuotedString, and it didn't look like the key had been corrupted at that point. I am willing to try to hunt this down if you can give me any suggestions on next steps.

bblanchon, 2022-03-11:
I'm afraid I cannot tell you what is wrong without looking at the complete program.
Please [contact me on codementor](https://www.codementor.io/@bblanchon) if you want me to review your code.

jpswensen, 2022-03-11:
I will work to come up with a complete example that still exhibits the problem (but don’t want to include my entire codebase because it is big and has far more than the issue at hand).

bblanchon, 2022-03-23:
@jpswensen, any progress?

closed 2022-04-06 by bblanchon