bblanchon/ArduinoJson #1728: multiple JSON

LEDZB, 2022-03-10:
I wonder if arduinojson can send multiple JSON data in one data sending?

bblanchon, 2022-03-10:
Hi @LEDZB,

I don't understand what you mean; please provide an example.

Best regards,
Benoit

LEDZB, 2022-03-10:
> Hi @LEDZB,
> 
> I don't understand what you mean; please provide an example.
> 
> Best regards, Benoit

For example, when I want to send JSON data again, first send {"food": "watermelon", "shelf life": "2 months", "quantity": 1, "purchase date": "sat, 21 Jul 2021 12:00:00 GMT"}, and then send another {"food": "apple", "shelf life": "3 months", "quantity": 3, "purchase date": "sat, 21 Jul 2021 12:00:00 GMT"}. I wonder if it can be realized with arduinojson?

bblanchon, 2022-03-11:
Yes, you can produce more than one JSON document with ArduinoJson.
To do that, you can either use two different `JsonDocument`s, or reuse the same one.
Please note that there are some caveats to reusing the same `JsonDocument` multiple times; see [I found a memory leak in the library!](https://arduinojson.org/v6/issues/memory-leak/)

closed 2022-03-23 by bblanchon