bblanchon/ArduinoJson #1729: Help needed with array parsing

cotestatnt, 2022-03-16:
Hi @bblanchon 
I am using your amazing library as a dependancy of mine and it's not very clear to me how I can solve a problem.

I have a JSON response from a remote server (Telegram) which varies little according to the command sent and I would like to find a smart way to manage this scenario.

Usually the answer is in this form:
`{"ok": true, "result": [{...}]}`
and I obviously need to parse the array **result** (only the first element really)

Unfortunately for some commands the answer is instead of this type:
`{"ok": true, "result": {...}}`

In your opinion, how can I handle this difference without rewriting the whole parser twice?

I was able without problems to determine if **result** is an array or not with a [simple test sketch](https://wokwi.com/projects/326301027495051858) like below.
Now, how can I use the content of result in the two different if/then/else block and then parse it as an object?

**Environment**  
Here is the environment that I'm using':
* Microconroller: ESP32
* Core/runtime: ESP32 core for Arduino v2.0.2
* IDE: Arduino IDE 1.8.16

**Reproduction**  
Here is a small snippet that demonstrate the problem.

```c++
char json[] = "{\"ok\": true,\"result\":{\"message_id\": 30214,\"from\": {\"id\": 123456789,\"is_bot\": true,\"first_name\": \"myFirstName\",\"username\": \"myUserName\"}}}";
//char json[] = "{\"ok\": true,\"result\":[{\"message_id\": 30214,\"from\": {\"id\": 123456789,\"is_bot\": true,\"first_name\": \"myFirstName\",\"username\": \"myUserName\"}}]}";
 
void setup() {
  Serial.begin(9600);
  StaticJsonDocument<512> doc;

  // Deserialize the JSON document
  DeserializationError error = deserializeJson(doc, json);  
  //serializeJsonPretty(doc, Serial); 

  if( doc["result"].as<JsonArray>().isNull() ) {    
    int id = doc["result"]["message_id"];
    Serial.println();    
    Serial.println(id);    
  } 
  else {    
    int id = doc["result"][0]["message_id"];
    Serial.println();    
    Serial.println(id);    
  }
```

bblanchon, 2022-03-17:
Hi @cotestatnt,

I think the best way to deal with this situation is to extract a [`JsonVariant`](https://arduinojson.org/v6/api/jsonvariant/) (or, better, a [`JsonVariantConst`](https://arduinojson.org/v6/api/jsonvariantconst/)), like so:

```c++
JsonVariant result = doc["result"];

if (result.is<JsonObject>()) {    
  int id = result["message_id"];
  // ...
} else if (result.is<JsonArray>()) {
  int id = result[0]["message_id"];
  // ...
}
```

Alternatively, you could point the `result` variable to the first element of the array, like so:

```c++
JsonVariant result = doc["result"];

if (result.is<JsonArray>())
  result = result[0];

int id = result["message_id"];
// ...
```

Best regards,
Benoit

cotestatnt, 2022-03-18:
Thanks @bblanchon 
Your second advice it's exactly what i was looking for and it works like a charm!

closed 2022-03-18 by cotestatnt