bblanchon/ArduinoJson #1730: Can't extract from nested JSON

nicksterious, 2022-03-18:
**Describe the bug**  
I am getting a compile error when trying to extract a portion of a nested JSON document.

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  
<details><summary>ArduinoJson Troubleshooter's report</summary> <ol><li>The issue happens at compile time</li><li>The error is not in the list</li></ol></details>

**Environment**  
Here is the environment that I used:
* Microcontroller: RP2040
* Core/runtime: arduino mbed 
* IDE: VSCode/platformio

**Reproduction**  
Here is a small snippet that reproduces the issue.

```JSON
{
  "state": {
    "desired": {
      "welcome": "aws-iot",
      "powerState": "OFF"
    },
    "reported": {
      "welcome": "aws-iot",
      "powerState": "OFF"
    }
  },
  "metadata": {
    "desired": {
      "welcome": {
        "timestamp": 1647421659
      },
      "powerState": {
        "timestamp": 1647594171
      }
    },
    "reported": {
      "welcome": {
        "timestamp": 1647421659
      },
      "powerState": {
        "timestamp": 1647594576
      }
    }
  },
  "version": 531,
  "timestamp": 1647620956
}
```

```c++
  DynamicJsonDocument incoming_message(2048);
  DynamicJsonDocument thing_state(2048);
  deserializeJson(incoming_message, incoming_message_str);
  thing_state = incoming_message["state"];
  // at this point thing_state should contain the "state" but not the "metadata"
```

**Compiler output**    
```
.pio\libdeps\pico\ArduinoJson\src/ArduinoJson/Document/BasicJsonDocument.hpp:100:31: error: 'const class ArduinoJson6193_F1::MemberProxy<ArduinoJson6193_F1::JsonDocument&, const char*>' has no member named 'memoryUsage'
  100 |     size_t requiredSize = src.memoryUsage();
      |                           ~~~~^~~~~~~~~~~
*** [.pio\build\pico\src\main.cpp.o] Error 1
```

Expected output:
Program should compile and maybe work.

bblanchon, 2022-03-19:
Hi @nicksterious,

Thank you very much for reporting this bug.
I'm going to fix it soon; meanwhile, please use the following workaround:

```c++
thing_state = incoming_message["state"].as<JsonVariant>();
```

Best regards,
Benoit

closed 2022-03-19 by bblanchon

bblanchon, 2022-04-05:
The fix was released with [ArduinoJson 6.19.4](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.19.4).

commit 3dc67c56 (2022-03-19): Add `memoryUsage()` to `ElementProxy` and `MemberProxy` (fixes #1730)