bblanchon/ArduinoJson #1731: Convert  Redis reply structure into a ArduinoJson document

vortex314, 2022-03-21:
I'm trying to convert a Redis reply in a ArduinoJson Document 
Both are highly recursive structures. In the recursion I thought about returning a JsonDocument as this is the real memory allocated. 
But the destructor of a JSON document is not accessible. 
Below the code that fails to compile on the line where I call replyToJson 

 error: ‘ArduinoJson6172_71::JsonDocument::~JsonDocument()’ is protected within this context 

The redisReply structure is described in : https://github.com/redis/hiredis/blob/master/hiredis.h
```
JsonDocument replyToJson(redisReply *reply) {
  DynamicJsonDocument doc(10240);
  switch (reply->type) {
    case REDIS_REPLY_STATUS:
    case REDIS_REPLY_ERROR:
    case REDIS_REPLY_BIGNUM:
    case REDIS_REPLY_VERB:
    case REDIS_REPLY_STRING: {
      doc = reply->str;
      break;
    };
    case REDIS_REPLY_DOUBLE: {
      doc=reply->dval;
      break;
    }
    case REDIS_REPLY_INTEGER: {
      doc=reply->integer;
      break;
    }
    case REDIS_REPLY_NIL: {
      doc=nullptr;
      break;
    }
    case REDIS_REPLY_BOOL: {
      doc=reply->integer != 0;
      break;
    }
    case REDIS_REPLY_MAP: {
      for (int i = 0; i < reply->elements; i += 2) {
          doc[reply->element[i]->str] = replyToJson(reply->element[i+1]);
      }
      break;
    }
    case REDIS_REPLY_SET:
    case REDIS_REPLY_PUSH:
    case REDIS_REPLY_ARRAY: {
      for (int i = 0; i < reply->elements; i++)
        doc.add( replyToJson(reply->element[i]);
      break;
    }
  }
  doc.shrinkToFit();
  return doc;
}
```

vortex314, 2022-03-21:
Ok this seems to work 
```
DynamicJsonDocument replyToJson(redisReply *reply) {
  DynamicJsonDocument doc(10240);
  
  switch (reply->type) {
    case REDIS_REPLY_STATUS:
    case REDIS_REPLY_ERROR:
    case REDIS_REPLY_BIGNUM:
    case REDIS_REPLY_VERB:
    case REDIS_REPLY_STRING: {
      doc.set(reply->str);
      break;
    };
    case REDIS_REPLY_DOUBLE: {
      doc.set(reply->dval);
      break;
    }
    case REDIS_REPLY_INTEGER: {
      doc.set(reply->integer);
      break;
    }
    case REDIS_REPLY_NIL: {
      doc.set(nullptr);
      break;
    }
    case REDIS_REPLY_BOOL: {
      doc.set(reply->integer != 0);
      break;
    }
    case REDIS_REPLY_MAP: {
      for (int i = 0; i < reply->elements; i += 2) {
          doc[reply->element[i]->str] = replyToJson(reply->element[i+1]);
      }
      break;
    }
    case REDIS_REPLY_SET:
    case REDIS_REPLY_PUSH:
    case REDIS_REPLY_ARRAY: {
      for (int i = 0; i < reply->elements; i++)
        doc.add( replyToJson(reply->element[i]));
      break;
    }
  }
  doc.shrinkToFit();
  return doc;
}
```

closed 2022-03-21 by vortex314

bblanchon, 2022-03-22:
Hi @vortex314,

Indeed, you cannot return a `JsonDocument` by value: you must use either a `StaticJsonDocument` or a `DynamicJsonDocument`.
You can, however, treat `StaticJsonDocument` and `DynamicJsonDocument` polymorphically by using a `JsonDocument&`, but that's not possible in your case.

The solution you found is sub-optimal because every recursive call causes a copy of the returned `DynamicJsonDocument`.
You could make this function many times faster by allocating the `DynamicJsonDocument` out of the recursion tree.
Here is an example:

```c++
DynamicJsonDocument replyToJson(redisReply *reply) {
  DynamicJsonDocument doc(10240);
  replyToJson(reply, doc.as<JsonVariant>());
  doc.shrinkToFit();
  return doc;
}

void replyToJson(redisReply *reply, JsonVariant result) {
  switch (reply->type) {
    case REDIS_REPLY_STATUS:
    case REDIS_REPLY_ERROR:
    case REDIS_REPLY_BIGNUM:
    case REDIS_REPLY_VERB:
    case REDIS_REPLY_STRING:
      result.set(reply->str);
      break;

    case REDIS_REPLY_DOUBLE:
      result.set(reply->dval);
      break;

    case REDIS_REPLY_INTEGER:
      result.set(reply->integer);
      break;

    case REDIS_REPLY_NIL:
      result.set(nullptr);
      break;

    case REDIS_REPLY_BOOL:
      result.set(reply->integer != 0);
      break;

    case REDIS_REPLY_MAP:
      for (int i = 0; i < reply->elements; i += 2)
        replyToJson(reply->element[i + 1], result[reply->element[i]->str]);
      break;

    case REDIS_REPLY_SET:
    case REDIS_REPLY_PUSH:
    case REDIS_REPLY_ARRAY:
      for (int i = 0; i < reply->elements; i++)
        replyToJson(reply->element[i], result.addElement());
      break;
  }
}
```

This code was obviously not tested, so it may contain some mistakes.

I would be very interested in knowing the difference in execution time.

Best regards,
Benoit

vortex314, 2022-03-22:
Hi Benoit, thanks for replying to a closed issue ! 
I'll have a look at your solution and provide some feedback if I can get it to work. 
The use case is that I want to enable IOT devices to use Redis functionality with a communication layer JSON based.
Similar to the serial2mqtt. https://github.com/vortex314/serial2mqtt
Keep you posted.
Lieven

vortex314, 2022-03-22:
Hi Benoit,
I adapted the code a bit to make it work. It looks 2x faster compared to my approach. Calling the function 100000 times fives 486 msec vs 911 msec.
Thanks
```
void replyToJson(JsonVariant result, redisReply *reply) {
  switch (reply->type) {
    case REDIS_REPLY_STATUS:
    case REDIS_REPLY_ERROR:
    case REDIS_REPLY_BIGNUM:
    case REDIS_REPLY_VERB:
    case REDIS_REPLY_STRING:
      result.set(reply->str);
      break;

    case REDIS_REPLY_DOUBLE:
      result.set(reply->dval);
      break;

    case REDIS_REPLY_INTEGER:
      result.set(reply->integer);
      break;

    case REDIS_REPLY_NIL:
      result.set(nullptr);
      break;

    case REDIS_REPLY_BOOL:
      result.set(reply->integer != 0);
      break;

    case REDIS_REPLY_MAP:
      for (int i = 0; i < reply->elements; i += 2)
        replyToJson(result[reply->element[i]->str].to<JsonVariant>(),
                    reply->element[i + 1]);
      break;

    case REDIS_REPLY_SET:
    case REDIS_REPLY_PUSH:
    case REDIS_REPLY_ARRAY:
      for (int i = 0; i < reply->elements; i++)
        replyToJson(result.addElement(), reply->element[i]);
      break;
  }
}

DynamicJsonDocument replyToJson(redisReply *reply) {
  DynamicJsonDocument doc(10240);
  replyToJson(doc.as<JsonVariant>(), reply);
  doc.shrinkToFit();
  return doc;
}
```