bblanchon/ArduinoJson #1734: How to get length of binary data string in JsonVariant

emielch, 2022-03-30:
I'm sending the contents of a binary file using ArduinoJson. The serializing and deserializing works all fine but I'm getting stuck at the point where I try to write the binary data back to a file.
Since the data can contain null characters, when trying to write it to a data file, the write function stops writing as soon as such a character occurs.
Therefore I would need to pass the length of the string to the write function so it doesn't stop on null characters.

```c++
int len = ??? // length of binary data string in JsonDoc["data"]
dataFile.write(JsonDoc["data"].as<const char*>(), len);
```

I know internally ArduinoJson knows the length, since printing the string using `serializeJson(JsonDoc["data"], Serial);` prints the string completely, with null characters encoded as "\u0000".

So my question is: is there a function that I could use, for example like `JsonDoc["data"].length()`, to extract the length from ArduinoJson?

Thanks in advance!

bblanchon, 2022-03-31:
Hi @emielch,

You can extract a [`JsonString`](https://arduinojson.org/v6/api/jsonstring/) (or an `std::string_view`) like so:

```c++
JsonString data = JsonDoc["data"];
dataFile.write(data.c_str(), data.size());
```

Best regards,
Benoit

closed 2022-04-13 by bblanchon

emielch, 2022-04-14:
Hey @bblanchon,
Sorry I forgot to thank you for your quick response, this was indeed what I was looking for!

Turns out that sending binary data this way doesn't work after all though. I was serializing it into MessagePack on a Node.js server and at first it seemed like it worked. But it turns out that according to the MessagePack spec, bytes in a binary string with a value above 127 (outside the ascii spec) gets preceded with some other byte. ArduinoJson doesn't recognize this spec, which causes those preceding bytes to be passed along when reading the binary string using the method you mentioned in your response, which corrupts the data.
Long story short, I just encode the data now into a Base64 string before serializing which avoids using non-ascii characters and decode that string into binary again on the arduino and everything works fine!

Thanks again for your help!