bblanchon/ArduinoJson #1738: Reuse jsonobject from another jsondocument

RensMvP, 2022-04-13:
**Describe the issue**  
I would like to use arduinoJson to serialize or deserialize data for my project.
The issue i am having is that i have several Json string that i need to send and some exist of duplicate data. For example i have 2 json string that have a jsonobject called device, but how do i reuse this object?

**Environment**  
I am using the Arduino IDE to program an ESP32.

**Example code**
In this ESP32 simulator is an example of my current inefficient setup, where i would like to reuse that device jsonobject.
[https://wokwi.com/projects/328805290179822162](url).

bblanchon, 2022-04-14:
Hi @RensMvP,

I'm sorry, I don't understand your question.
Can you rephrase?

Best regards,
Benoit

RensMvP, 2022-04-14:
In my program i have 2 json strings that i send with both a jsonobject device. Both json strings make use of the same data for the jsonobject device.

In my current code i have 2 json object called device and i was wondering if it is possible to have a function that return the device object to be used in the json string that i return to the main program.

``` C++
String JsonData::jsonPairRequest()
{
    DynamicJsonDocument doc(1024);
    String rtnJson;

    // Json used for pair/request
    JsonArray scope = doc.createNestedArray("scope");
    scope.add("read");
    scope.add("write");
    scope.add("control");

    // json object device
    JsonObject device = doc.createNestedObject("device");
    device["device_name"] = "heliotrope";
    device["device_os"] = "Android";
    device["app_name"] = "ESP32api";
    device["type"] = "native";
    device["app_id"] = "app.id";
    device["id"] = "mMAkqVsvIE5s2nw4"; 

    serializeJson(doc, rtnJson);   // Convert the Json data to String
    return rtnJson;    
}
    
````

Another function with the same device but in another json document
``` C++
String JsonData::jsonGrantPair(int pinCode, int timeStamp, const char* signature)
{
    DynamicJsonDocument doc(1024);
    String rtnJson;

    // Json used for the auth, pair/grant
    JsonObject auth = doc.createNestedObject("auth");
    auth["auth_AppId"] = "1";
    auth["pin"] = pinCode;
    auth["auth_timestamp"] = timeStamp;
    auth["auth_signature"] = signature;

    // another json object device (same as the other one)
    JsonObject deviceGrant = doc.createNestedObject("device");
    deviceGrant["device_name"] = "heliotrope";
    deviceGrant["device_os"] = "Android";
    deviceGrant["app_name"] = "ESP32api";
    deviceGrant["type"] = "native";
    deviceGrant["app_id"] = "app.id";
    deviceGrant["id"] = "mMAkqVsvIE5s2nw4";

    serializeJson(doc, rtnJson);   // Convert the Json data to String
    return rtnJson;    
}```

bblanchon, 2022-04-15:
If I understand correctly, your question is "how can I create a function that fills the "device" member".

There are three options:

1. Return a `JsonDocument`
2. Pass a `JsonObject` to the function
3. Define custom converter

In the following code samples, I'm going to assume that you have a struct like this:

```c++
struct Device {
  char name[32];
  char os[32];
  char appName[32];
  char type[16];
  char appId[32];
  char id[64];
};
```

### Option 1: Return a `JsonDocument`

```c++
StaticJsonDocument<128> makeDeviceJson(const Device& device) {
  StaticJsonDocument<128> doc;
  doc["device_name"] = device.name;
  doc["device_os"] = device.os;
  doc["app_name"] = device.appName;
  doc["type"] = device.type;
  doc["app_id"] = device.appId;
  doc["id"] = device.id;
  return doc;
}
```

You use this function like so:

```c++
doc["device"] = makeDeviceJson(device);
```

Pros:

* Calling code looks really nice

Cons:

* It makes a copy of the JSON object

### Option 2: Pass a `JsonObject`

```c++
void fillDeviceJson(const Device& device, JsonObject obj) { 
  obj["device_name"] = device.name;
  obj["device_os"] = device.os;
  obj["app_name"] = device.appName;
  obj["type"] = device.type;
  obj["app_id"] = device.appId;
  obj["id"] = device.id;
}
```

You use this function like so:

```c++
fillDeviceJson(device, doc.createNestedObject("device"));
```

Pros:

* No copy

Cons:

* Calling code is less elegant

### Option 3: Custom converter

Please read the [documentation for customer converters](https://arduinojson.org/news/2021/05/04/version-6-18-0/)

```c++
void convertToJson(const Device& src, JsonVariant dst) { 
  dst["device_name"] = src.name;
  dst["device_os"] = src.os;
  dst["app_name"] = src.appName;
  dst["type"] = src.type;
  dst["app_id"] = src.appId;
  dst["id"] = src.id;
}
```

You use this function like so:

```c++
doc["device"] = device;
```

Pros:

* No copy
* Calling code kicks ass

Cons:

* Harder to see what's happening

RensMvP, 2022-04-22:
Sorry for my late response, but awesome options! Must admit that i did not expect suchs and wel explained piece of code.
I think that solution 3 might be the nices to have in my code and easiest to use.

bblanchon, 2022-04-23:
You're welcome, @RensMvP.
Please close the issue as soon as you confirmed that the solution worked.

closed 2022-05-02 by bblanchon