bblanchon/ArduinoJson #1739: The Key of a first key:value pairs disappears with no code affecting it

snowfrogblues, 2022-04-13:
HI there; 
I've gone through the trouble shooter but as you will see there are no valid selection for what I see. 
I trying to convert a piece of code that convert each elements of a json config file into discrete but global variables into something where I can use the actual json document as a global. I got some strange behavior which brought me here. I'll show the strangeness below. 

First this is for PIO
This is my .ini
```
[env:esp01_1m]
platform = espressif8266
board = esp01_1m
framework = arduino
;upload_port = com[3]
monitor_speed = 115200
board_build.flash_mode = dout

lib_deps = 
        tzapu/WiFiManager @ ^0.16.0
        bblanchon/ArduinoJson @ ^6.19.3
        beegee-tokyo/DHT sensor library for ESPx @ ^1.18
        256dpi/MQTT @ ^2.5.0
monitor_filters = esp8266_exception_decoder, default

#board_build.filesystem = littlefs
```
The global declaration of the jsonConfig is:
         ` StaticJsonDocument<512>      jsonConfig;`
but I also tried DynamicJsonDocument with no change. 

The code fragment is taken out of a readConfig() function that reads from a SPIFFS file 
-----------------------------------------------------------------------------------------------
Focus on all things "mqtt_server". The Key of a key:value pair disappears!!!

```
        auto deserializeError = deserializeJson(jsonConfig, buf.get());
        serializeJson(jsonConfig, Serial);                   //-1 first json in the output below - all is good but not for long
        if ( ! deserializeError ) {
          Serial.println("\nExtract json");

          strcpy(ip, jsonConfig[IP]);                                      //This works and stays valid throughout
          strcpy(mqtt_server, jsonConfig[MQTT_SERVER]);                    // This works and is also used throughout
          strcpy(mqtt_port, jsonConfig[MQTT_PORT]);
          strcpy(mqtt_token, jsonConfig[MQTT_TOKEN]);
          strcpy(device_name, jsonConfig[DEVICE_NAME]);
          strcpy(topic_prefix, jsonConfig[TOPIC_PREFIX]);
          strcpy(isdeviceonBatt, jsonConfig[ISDEVONBATT]);

          Serial.print("The value of MQTT_SERVER is : ");
          Serial.println(String(jsonConfig[MQTT_SERVER]));               // -2 This works but same statement below will return null ????
                                                                   
        } else {
            //deserialize error
          Serial.println("failed to load json content");
          return false;
        }
      } else {
          Serial.println("Could not open config.json");
          return false;
      }
    } else {
        Serial.println("config.json not found");
        return false;
    }
  } else {
    Serial.println("failed to mount FS");
    return false;
  }
  // return true mean all went good
  Serial.println("Config retrieved");
  Serial.print("On exiting for readConfig The value od MQTT_SERVER is : ");
          Serial.println(String(jsonConfig[MQTT_SERVER]));       //exact same statement as above (-2) but this returns null ????
          Serial.println("Serialize on exit of readconfig");
          serializeJson(jsonConfig, Serial);                //exact same statement as above (-1) but note in the console below
                                                          // the "mqtt_server":"10.0.0.70"..... is now "":"10.0.0.70"
          Serial.println("\n\n\n");
  return true;
}
```
The trace on the console for the following fragment is 
-----------------------------------------------------------
```
{"mqtt_server":"10.0.0.70","mqtt_port":"8883","mqtt_token":"Mqtt-security-token","device_name":"Prototype-IOT","topic_prefix":"/house/indoor/office/","is_device_on_battery":"N","ip":"10.0.0.207","gateway":"10.0.0.1","subnet":"255.255.255.0"}

Extract json

The value of MQTT_SERVER is : 10.0.0.70
Config retrieved
On exiting for readConfig The value od MQTT_SERVER is : null
Serialize on exit of readconfig
{"":"10.0.0.70","mqtt_port":"8883","mqtt_token":"Mqtt-security-token","device_name":"Prototype-IOT","topic_prefix":"/house/indoor/office/","is_device_on_battery":"N","ip":"10.0.0.207","gateway":"10.0.0.1","subnet":"255.255.255.0"}
```

So all appears to work but in the span of a few "debug" lines the first element of the json document looses its key?
I could not find a bug or issues that relates to that. 

Does anyone have any ideas?

bblanchon, 2022-04-14:
Hi @snowfrogblues,

Can you share the Troubleshooter's report?

Best regards,
Benoit

snowfrogblues, 2022-04-14:
I spend much more time trying different variations with the trouble shooter and finally found one that gave me the solution. 
THe reason I was not very successful with the troubleshooter before was that I did assume that the issue was with the deserialization but using that path lead nowere. So I switched to serialization and one of the question was "Do you call [deserializeJson()] before calling [serializeJson()]? (with the same [JsonDocument]" and that lead me to the solution as below

Merci Benoit

Answer from troubleshooter. 
When you pass a writeable buffer as the input of [deserializeJson()], ArduinoJson uses the zero-copy mode. Instead of copying the strings from the input into the [JsonDocument], it stores pointers to the input buffer. Therefore, the input buffer must stay alive during the whole lifetime of the [JsonDocument]; otherwise, all the strings in the [JsonDocument] will be dangling pointers. Usually, this produces grabage in the output, but it can sometimes produce empty strings.

The easiest solution is to disable the zero-copy mode by passing a read-only input buffer. Casting the pointer will do the trick:

// replace this:
deserializeJson(doc, input, inputSize);

// ...with this:
deserializeJson(doc, (const char*)input, inputSize);
Now that the zero-copy mode is disabled, you probably need a larger [JsonDocument], so don’t forget to update the capacity. As usual, use the [ArduinoJson Assistant] to compute the right capacity for your project.

closed 2022-04-14 by snowfrogblues

bblanchon, 2022-04-14:
You're welcome, @snowfrogblues; thank you for the update
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink:
Good luck with your project!