bblanchon/ArduinoJson #1740: How to create CustomWriter to serialize SpiRamJsonDocument into PubSubClient.write()?

iwas108, 2022-04-15:
**Describe the issue**  
I have a SpiRamJsonDocument doc, and need to send the content (Serialize)  to PubSubClient.write() that accept a one by one byte or by byte buffer. I have try to use a CustomWriter, but have no luck and don't know how to use it. The data is very large, it is a base64 image that is stored on PSRAM, and I need to send it via MQTT.

**Environment**  
Here is the environment that I'm using':
* Microconroller: [e.g. ESP32CAM AI Thinker]
* Core/runtime: [e.g. ESP32 core for Arduino Latest Version]
* IDE: [e.g. PlatformIO Latest Version]

**Reproduction**  
Here is a small snippet that demonstrate the problem.

```c++
void myTask()
{
  //if(tb.connected())
  if(true)
  {
    camera_fb_t * fb = NULL;
    fb = esp_camera_fb_get();

    if (!fb) {
      sprintf_P(logBuff, PSTR("Camera capture failed."));
      recordLog(1, PSTR(__FILE__), __LINE__, PSTR(__func__));
    }
    else
    {
      size_t size = base64_encode_expected_len(fb->len) + 1;
      char * buffer = (char *) ps_malloc(size);
      if(buffer) {
        base64_encodestate _state;
        base64_init_encodestate(&_state);
        int len = base64_encode_block((const char *) &fb->buf[0], fb->len, &buffer[0], &_state);
        len = base64_encode_blockend((buffer + len), &_state);
      }
      esp_camera_fb_return(fb);

      SpiRamJsonDocument doc(size + 64);
      doc["b64"] = String(buffer).c_str();
      free(buffer);
      doc["w"] = fb->width;
      doc["h"] = fb->height;
      doc["f"] = fb->format;

      size_t finalDataSize = measureJson(doc);
      char * data = (char *) ps_malloc(finalDataSize);
      serializeJson(doc, data, finalDataSize);
      doc.clear();

      tb.beginPublish("v1/devices/me/telemetry", finalDataSize, 0);
      for(int i=0;i<finalDataSize;i++){
        float progress = ((float)i+1.0) / (float)(finalDataSize) * 100.0;
        sprintf_P(logBuff, PSTR("Sending chunk of captured data %d/%d: (%.2f %%)"), i+1, finalDataSize, progress);
        Serial.print(data[i]);
        Serial.print(" - ");
        Serial.println(logBuff);
        tb.write(data[i]);
      }
      tb.endPublish();
    }
  }
}
// insert code here
```

Thankyou, I'hope someone can help on the for loop part to send the data by chunk of buffer. Or implement CustomWriter to PubSub.write().

bblanchon, 2022-04-17:
Hi,

As far as I know, you cannot.
Please see [How to use ArduinoJson with PubSubClient?](https://arduinojson.org/v6/how-to/use-arduinojson-with-pubsubclient/).

I strongly recommend **against** using JSON to transmit images, and PubSubClient limits message size to 256 bytes by default.

Best regards,
Benoit

iwas108, 2022-04-20:
Hi Benoit, thanks for the recommendation. Finally I found the solution, this is special case due to IoT infrastructure that only accept communication via MQTT, so I must implement an image capture function and send the image in base64 thru MQTT. This solution of course imperfect and inefficient, but I'll post here so anyone have same case can benefit.

```
void myTask()
{
  if(tb.connected())
  {
    camera_config_t cam;
    cam.ledc_channel = LEDC_CHANNEL_0;
    cam.ledc_timer = LEDC_TIMER_0;
    cam.pin_d0 = Y2_GPIO_NUM;
    cam.pin_d1 = Y3_GPIO_NUM;
    cam.pin_d2 = Y4_GPIO_NUM;
    cam.pin_d3 = Y5_GPIO_NUM;
    cam.pin_d4 = Y6_GPIO_NUM;
    cam.pin_d5 = Y7_GPIO_NUM;
    cam.pin_d6 = Y8_GPIO_NUM;
    cam.pin_d7 = Y9_GPIO_NUM;
    cam.pin_xclk = XCLK_GPIO_NUM;
    cam.pin_pclk = PCLK_GPIO_NUM;
    cam.pin_vsync = VSYNC_GPIO_NUM;
    cam.pin_href = HREF_GPIO_NUM;
    cam.pin_sscb_sda = SIOD_GPIO_NUM;
    cam.pin_sscb_scl = SIOC_GPIO_NUM;
    cam.pin_pwdn = PWDN_GPIO_NUM;
    cam.pin_reset = RESET_GPIO_NUM;
    cam.xclk_freq_hz = 20000000;
    cam.pixel_format = PIXFORMAT_JPEG;
    cam.fb_count = 2;
    cam.frame_size = (framesize_t)mySettings.frameSize;
    cam.jpeg_quality = mySettings.jpegQuality;

    sprintf_P(logBuff, PSTR("Taking snap (jpegQuality: %d, frameSize: %d), please wait."), cam.jpeg_quality, (int)cam.frame_size);
    recordLog(1, PSTR(__FILE__), __LINE__, PSTR(__func__));

    // camera init
    esp_err_t err = esp_camera_init(&cam);
    if (err != ESP_OK) {
      sprintf_P(logBuff, PSTR("Camera init failed with error 0x%x"), err);
      recordLog(1, PSTR(__FILE__), __LINE__, PSTR(__func__));
      esp_camera_deinit();
    }
    delay(5000);
    camera_fb_t * fb = NULL;
    fb = esp_camera_fb_get();

    if (!fb) {
      sprintf_P(logBuff, PSTR("Camera capture failed."));
      recordLog(1, PSTR(__FILE__), __LINE__, PSTR(__func__));
    }
    else
    {
      size_t size = base64_encode_expected_len(fb->len) + 1;
      char * buffer = (char *) ps_malloc(size);
      if(buffer) {
        base64_encodestate _state;
        base64_init_encodestate(&_state);
        int len = base64_encode_block((const char *) &fb->buf[0], fb->len, &buffer[0], &_state);
        len = base64_encode_blockend((buffer + len), &_state);
      }
      esp_camera_fb_return(fb);
      esp_camera_deinit();

      SpiRamJsonDocument doc(size + 64);
      doc["b64"] = buffer;
      free(buffer);
      doc["w"] = fb->width;
      doc["h"] = fb->height;
      doc["f"] = fb->format;

      size_t finalDataSize = measureJson(doc);
      byte * data = (byte *) ps_malloc(finalDataSize);
      serializeJson(doc, data, finalDataSize);
      doc.clear();

      sprintf_P(logBuff, PSTR("Snap size %d.  Memory: %d/%d - %d. Sending now..."), finalDataSize, heap_caps_get_free_size(MALLOC_CAP_32BIT), ESP.getPsramSize(), ESP.getFreeHeap());
      recordLog(1, PSTR(__FILE__), __LINE__, PSTR(__func__));

      const char* topic = "v1/devices/me/telemetry";
      publishMqtt(topic, data, finalDataSize);
      free(data);
    }
  }
}

void publishMqtt(const char* channel, uint8_t *data, uint32_t len) {
  unsigned long start_ts = millis();

  tb.beginPublish(channel, len, false);

  size_t res;
  uint32_t offset = 0;
  uint32_t to_write = len;
  uint32_t buf_len;
  do {
    buf_len = to_write;
    if (buf_len > mySettings.tempBuffSize){
      buf_len = mySettings.tempBuffSize;
    }
    res = tb.write(data+offset, buf_len);
    offset += buf_len;
    to_write -= buf_len;
  } while (res == buf_len && to_write > 0);

  tb.endPublish();

  sprintf_P(logBuff, PSTR("Finished: (size %d bytes, %d bytes written in %ld ms)\n"), len, len-to_write, millis()-start_ts);
  recordLog(1, PSTR(__FILE__), __LINE__, PSTR(__func__));
}
```

Here, I can send up to 512KB of base64 image via MQTT using PubSubClient (AI Thinker ESP32-Cam), with little modification on PubSubClient max header size, from uint16_t to uint32_t. Hope this help someone else who facing the same problem.

closed 2022-04-20 by iwas108