bblanchon/ArduinoJson #1741: DeserializationError::Ok even if input is not a JSON, but just an integer

ishdemon, 2022-04-16:
Input is an integer 7, and still it says DeserializationError: OK. How to throw error or handle if its not JSON. Shouldn't it say InvalidInput or something??

bblanchon, 2022-04-17:
Hi @ishdemon,

As I explained in https://github.com/bblanchon/ArduinoJson/issues/1575#issuecomment-850786675:

> A number is a valid JSON document.
> You can do the same in your web browser:
>
> ```
> JSON.parse('7')
> < 7
> ```

The [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/#runtime/deserialization/ok/invalid) seems to agree with me when he says:

> You know, a JSON document doesn’t need to start with a { or a [; the following strings are also valid JSON documents:
> 
> * `0`
> * `1`
> * `-9`
> * `1.234e-34`
> * `true`
> * `false`
> * `null`

If you want to verify that the JSON document is an object, simply call [JsonDocument::is<JsonObject>()](https://arduinojson.org/v6/api/jsondocument/is/).

Best regards,
Benoit

ishdemon, 2022-04-17:
Thanks. Not aware of this.. Tried looking closed issues but couldn't find. 
```
if (doc.is<int>()) {
  int value = doc.as<int>();
  ...
}
```
Closing this. Thanks again🙏👍

closed 2022-04-17 by ishdemon