bblanchon/ArduinoJson #1743: ArduinoJSON and ESP32

basementmedia2, 2022-04-18:
Hi,

i want to use ArduinoJSON with ESP32.
Is there anything to consider when handling the library, since the ESP32 has a reduced StackSize in the loop task?
Also a question about the Assistant (https://arduinojson.org/v6/assistant/):
In Step 2 of the Assistant i have put in this:
```
[
  {
    "s": 1,
    "gk": "90440",
    "l": 133
  },
  {
    "s": 2,
    "gk": "27200",
    "l": 40
  },
  {
    "s": 3,
    "gk": "0",
    "l": 0
  },
  {
    "s": 4,
    "gk": "7480",
    "l": 11
  },
  {
    "s": 5,
    "gk": "92480",
    "l": 136
  },
  {
    "s": 6,
    "gk": "187680",
    "l": 276
  },
  {
    "s": 7,
    "gk": "158440",
    "l": 233
  },
  {
    "s": 8,
    "gk": "244120",
    "l": 359
  },
  {
    "s": 9,
    "gk": "329120",
    "l": 484
  },
  {
    "s": 10,
    "gk": "336600",
    "l": 495
  },
  {
    "s": 11,
    "gk": "447440",
    "l": 658
  },
  {
    "s": 12,
    "gk": "286280",
    "l": 421
  },
  {
    "s": 13,
    "gk": "230520",
    "l": 339
  },
  {
    "s": 14,
    "gk": "218280",
    "l": 321
  },
  {
    "s": 15,
    "gk": "257720",
    "l": 379
  },
  {
    "s": 16,
    "gk": "416160",
    "l": 612
  },
  {
    "s": 17,
    "gk": "224839",
    "l": 333.9199829
  },
  {
    "s": 18,
    "gk": "448453",
    "l": 672.1199951
  },
  {
    "s": 19,
    "gk": "759560",
    "l": 1117
  },
  {
    "s": 20,
    "gk": "584800",
    "l": 860
  },
  {
    "s": 21,
    "gk": "278120",
    "l": 409
  },
  {
    "s": 22,
    "gk": "98600",
    "l": 145
  },
  {
    "s": 23,
    "gk": "177480",
    "l": 261
  }
]
```

Step 4 then suggested to us

StaticJsonDocument<2048> doc;

But documentation says that when stacksize is more than 1KB you should use DynamicJsonDocument.
In this case i have 2048 Bytes but Assistant says StaticJsonDocument?

Or is 2048 not bytes?

Sorry for this beginner question.

Best wishes and thank you for this great library!

Best wishes
Daniel

bblanchon, 2022-04-19:
Hi Daniel,

As far as I know, the [ESP32 core for Arduino](https://github.com/espressif/arduino-esp32) defines a default stack size of 8KB
The ArduinoJson Assistant knows that and uses a `DynamicJsonDocument` for everything above 2KB.

A `StaticJsonDocument<2048>` should work fine on an ESP32, except if you have other large variables in the stack.

Best regards,
Benoit

basementmedia2, 2022-04-21:
Omg thats a shame ;-(
Sure 2 kb are 2048...sorry.

Best regards

Daniel Bäuerlein

closed 2022-04-21 by bblanchon