bblanchon/ArduinoJson #1745: Error when deserializing

MaralS, 2022-04-20:
**Error when deserializing json data from graph.facebook.com**  
I'm trying to deserialized the data that I extracted from the Facebook API.
```
17:35:52.005 -> ======================================
17:35:52.005 -> ===========GET FB FAN COUNT===========
17:35:52.005 -> ======================================
17:35:52.005 -> graph.facebook.com
17:35:53.025 -> HTTPS Connecting : graph.facebook.com
17:35:54.329 -> Connected to web
17:35:54.375 -> Request sent
17:35:54.468 -> Headers received
17:35:54.468 -> Request result :{"fan_count":447,"id":"ABCDEFG"}
17:35:57.442 -> Deserialization process
17:36:00.470 -> deserializeJson() failed: EmptyInput
17:36:00.470 -> ======================================
17:36:00.470 -> ===========GET IG FOLLOWERS===========
17:36:00.470 -> ======================================
17:36:00.470 -> graph.facebook.com
17:36:01.448 -> HTTPS Connecting : graph.facebook.com
17:36:02.747 -> Connected to web
17:36:02.794 -> Request sent
17:36:02.932 -> Headers received
17:36:02.932 -> Request result :{"followers_count":94,"id":"ABCDEFG"}
17:36:05.952 -> Deserialization process
17:36:08.928 -> deserializeJson() failed: EmptyInput
 
```
**Troubleshooter report**  

**Environment**  
Here is the environment that I'm using':
* Microconroller: [e.g. ESP8266] - Wemos D1 mini
* Core/runtime: [e.g. ESP8266 core for Arduino v1.8.19]
* IDE: [e.g. Arduino IDE 1.8.16]

**Reproduction**  
Here is a small snippet that demonstrate the problem.

```
//URL PATTERN /AAA?access_token=BBB&fields=CCC
//AAA - fb_id_nb ou ig_id_nb
//BBB - fb_long_access_token
//CCC - fan_count (FB) ou followers_count (IG)

const String fb_fans = "fan_count";
const String ig_followers = "followers_count";

const char* host = "graph.facebook.com";
const int httpsPort = 443;

const String facebookUrl = "/" + fb_id_nb + "?access_token=" + fb_long_access_token + "&fields=" + fb_fans + "";
const String instagramUrl = "/" + ig_id_nb + "?access_token=" + fb_long_access_token + "&fields=" + ig_followers + "";

void connectToWifi() {
  WiFi.hostname("Social Counter Device (WEMOS D1 mini)");
  WiFi.begin(ssid, pwd);
  Serial.println("======================================");
  Serial.println("============WIFI CONNECTION===========");
  Serial.println("======================================");
  Serial.print("Connecting");
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println();
  Serial.print("Connected, IP address: ");
  Serial.println(WiFi.localIP());
}
void getSocialCount(String url) {
  WiFiClientSecure httpsClient;    //Declare object of class WiFiClient
  Serial.println("======================================");

  if (url != facebookUrl) {
    Serial.println("===========GET IG FOLLOWERS===========");

  } else {
    Serial.println("===========GET FB FAN COUNT===========");
  }
  Serial.println("======================================");
  Serial.println(host);
  httpsClient.setInsecure(); // permet de se passer du fingerprint du certificat SSL
  httpsClient.setTimeout(3000); // 3 s sec avant d'afficher la requete
  delay(1000);
  Serial.print("HTTPS Connecting : ");
  Serial.println(host);
  int r = 0; //retry counter
  while ((!httpsClient.connect(host, httpsPort)) && (r < 30)) {
    delay(100);
    Serial.print(".");
    r++;
  }
  if (r == 30) {
    Serial.println("Connection failed");
  }
  else {
    Serial.println("Connected to web");
  }
  httpsClient.print(String("GET ") + url + " HTTP/1.1\r\n" +
                    "Host: " + host + "\r\n" +
                    "Connection: close\r\n\r\n");
  Serial.println("Request sent");
  while (httpsClient.connected()) {
    String line = httpsClient.readStringUntil('\n');
    if (line == "\r") {
      Serial.println("Headers received");
      break;
    }
  }
  Serial.print("Request result :");
  String line;
  while (httpsClient.available()) {
    line = httpsClient.readStringUntil('\n');  //Read Line by Line
    Serial.println(line); //Print response
  }
  Serial.println("Deserialization process");
  
  DynamicJsonDocument doc(96); //For ESP32/ESP8266 you'll mainly use dynamic.
  Serial.setTimeout(10000);
  DeserializationError error = deserializeJson(doc, httpsClient);
  if (!error) {
    if (url != instagramUrl) {
      float fan_count = doc["fan_count"]; //FB  
      Serial.print("Facebook Fans : ");
      Serial.println(fan_count);
    } else {
      float followers_count = doc["followers_count"];//IG
      Serial.print("Instagram followers : ");
      Serial.println(followers_count);
    }
  } else {
    Serial.print(F("deserializeJson() failed: "));
    Serial.println(error.f_str());
    return;
  }
  Serial.println("Closing connection");
}
void setup() {
  Serial.begin(74880);
  delay(500);
  connectToWifi();
}
void loop() {
  if ((WiFi.status() == WL_CONNECTED)) {
    getSocialCount(facebookUrl);
    getSocialCount(instagramUrl);

  }
  delay(180000);//3 min de delay
}
```

**Program output**  
```
17:35:52.005 -> ======================================
17:35:52.005 -> ===========GET FB FAN COUNT===========
17:35:52.005 -> ======================================
17:35:52.005 -> graph.facebook.com
17:35:53.025 -> HTTPS Connecting : graph.facebook.com
17:35:54.329 -> Connected to web
17:35:54.375 -> Request sent
17:35:54.468 -> Headers received
17:35:54.468 -> Request result :{"fan_count":447,"id":"ABCDEFG"}
17:35:57.442 -> Deserialization process
17:36:00.470 -> deserializeJson() failed: EmptyInput
17:36:00.470 -> ======================================
17:36:00.470 -> ===========GET IG FOLLOWERS===========
17:36:00.470 -> ======================================
17:36:00.470 -> graph.facebook.com
17:36:01.448 -> HTTPS Connecting : graph.facebook.com
17:36:02.747 -> Connected to web
17:36:02.794 -> Request sent
17:36:02.932 -> Headers received
17:36:02.932 -> Request result :{"followers_count":94,"id":"ABCDEFG"}
17:36:05.952 -> Deserialization process
17:36:08.928 -> deserializeJson() failed: EmptyInput
 
```

I tried to put a delay, a timeout, ..it's probably an error from me, I'm new to the C++ universe.

Thanks a lot for this amazing library !

closed 2022-04-21 by MaralS

bblanchon, 2022-04-21:
Hi @MaralS,

Indeed, you have to choose one approach:

* either read the body and pass it to `deserializeJson()`,
* or pass the `Client` to `deserializeJson()` so that it reads the body itself.

But you cannot do both at the same time.

Best regards,
Benoit

bblanchon, 2022-04-21:
PS: consider using `HTTP/1.0` instead of `HTTP/1.1`  to avoid [Chunked Transfer Encoding](https://en.wikipedia.org/wiki/Chunked_transfer_encoding).

MaralS, 2022-04-22:
@bblanchon thanks a lot for the information. I bought your book because your library is awesome ! The best way to support the hardwork

bblanchon, 2022-04-23:
Thank you very much for your support, @MaralS; it means a lot!