bblanchon/ArduinoJson #1749: "test" string slipping through deserializeJson(), imitate as valid JSON string input.

OptifySudarshanPatil, 2022-05-07:
**Describe the issue**  
Sending "test" to arduino serial which has to be generate "InvalidInput" error.

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  
1. The issue happens at run time
2. The issue concerns deserialization
3. <code>deserializeJson()</code> returns <code>Ok</code>
4. <code>Ok</code> is incorrect, it should be <code>InvalidInput</code>
5. jsonlint says the document is invalid
6. <code>deserializeJson()</code> should not let this error pass

**Environment**  
Here is the environment that I'm using':
* Microconroller: [e.g. Arduino DUE]
* Core/runtime: [e.g. Atmel SAM 7.1.0]
* IDE: [e.g. PlatformIO Core, version 5.2.5]

**Reproduction**  
Here is a small snippet that demonstrate the problem.

```c++
#include <Arduino.h>
#include <ArduinoJson.h>
void setup()
{
  // put your setup code here, to run once:

  Serial.begin(115200);
}
void loop()
{
  if (Serial.available() > 0)
  {
    in_data = Serial.readStringUntil('\n');
    // decode_json(&in_data);
    char nodered_in_data[20000];
    StaticJsonDocument<20000> nodered_in;

    uint32_t in_msg_length = in_data.length() + 1;

    String in_data.toCharArray(nodered_in_data, in_msg_length);
    DeserializationError error = deserializeJson(nodered_in, nodered_in_data);
    if (error)
    {
      Serial.print(F("deserializeJson() failed: "));
      Serial.println(error.f_str());
      Serial.println(nodered_in_data);
      return;
    }
    else
    {
      Serial.print("JSON Received: ");
      Serial.println(nodered_in_data);
    }

    JsonVariant json_receive = nodered_in["time_rx"];
    if (!json_receive.isNull())
    {
      String time_rx = json_receive;
      Serial.print("time_rx ");
      Serial.println(time_rx);
    }

    JsonVariant json_blink_time = nodered_in["blink_time"];
    if (!json_blink_time.isNull())
    {
      String blink_time = json_blink_time;
      Serial.print("blink_time ");
      Serial.println(blink_time);
    }
  }
}
```

**Program output**  
If relevant, include the program output.

Expected output:

```
test
deserializeJson() failed: InvalidInput
test
```

Actual output:

```
test
JSON Received: test
{"blink_time":"500"}
JSON Received: blink_time
blink_time 500
{"time_rx":"abc"}
JSON Received: time_rx
time_rx abc
```
[image]

bblanchon, 2022-05-07:
Hi @OptifySudarshanPatil,

As the ArduinoJson Troubleshooter already said:

> ArduinoJson isn't very picky about the input: **its implementation favors code size and speed over strict conformance**.
> After all, size and speed are what matter the most for embedded software, right?
> 
> This means that ArduinoJson's parser may accept documents that would be rejected by other parsers.
> 
> For example, `deserializeJson()` tolerates the following substitutions:
> 
> * `'hello'` instead of `"hello"` (single quotes)
> * `nULL` or `n0n3` instead if `null` (only checks first character and length)
> * `tRUE` or `t0t0` instead of `true` (ditto)
> * `fALSE` or `fAkk3` instead of `false` (ditto)

In your case, `test` (4 characters starting with a `t`) is interpreted as `true`.

I don't think it's worth increasing the size of the library for that.

Best regards,
Benoit

OptifySudarshanPatil, 2022-05-09:
Thanks @bblanchon,
This was a very usefull input.
i will avoid [this ](https://github.com/bblanchon/ArduinoJson/issues/1749#:~:text=%27hello%27%20instead,false%20(ditto))at all cost now.

closed 2022-05-09 by OptifySudarshanPatil