bblanchon/ArduinoJson #1750: Converting back ArduinoJsonAssistantv6 generated code to json is incomplete

rjjrbatarao, 2022-05-07:
**Describe the bug**  
I was trying to serialize a large json object (like below)  from the output of ArduinoJsonv6 assistant, I copied the generated c++ code and pasted as json response of my webserver   using this WiFiClient client = server.client();   serializeJson(doc, client);, but json was incomplete I was expecting to get back below json object.

``` json
[
   {
      "e": {
         "t": "nav",
         "a": {
            "class": [
               "navbar",
               "navbar-expand-md",
               "navbar-dark",
               "bg-dark",
               "fixed-top"
            ],
            "id": "nav_1"
         },
         "b": [
            {
               "e": {
                  "t": "a",
                  "p": "nav_1",
                  "a": {
                     "class": [
                        "navbar-brand"
                     ],
                     "href": "#"
                  },
                  "b": "Navbar"
               }
            },
            {
               "e": {
                  "t": "button",
                  "p": "nav_1",
                  "a": {
                     "class": [
                        "navbar-toggler"
                     ],
                     "type": "button",
                     "data_toggle": "collapse",
                     "data_target": "#navbarsExampleDefault",
                     "aria_controls": "navbarsExampleDefault",
                     "aria_expanded": "false",
                     "aria_label": "Toggle navigation",
                     "id": "btn_1",
                     "href": "#"
                  },
                  "b": {
                     "e": {
                        "t": "span",
                        "a": {
                           "class": [
                              "navbar-toggler-icon"
                           ]
                        },
                        "b": ""
                     }
                  }
               }
            },
            {
               "e": {
                  "t": "div",
                  "p": "nav_1",
                  "a": {
                     "class": [
                        "collapse",
                        "navbar-collapse"
[... 747 lines cut ...]
                     },
                     {
                        "e": {
                           "t": "p",
                           "p": "str_id",
                           "a": {
                              "class": "lead"
                           },
                           "b": "Use this document as a way to quickly start any new project.\n All you get is this text and a mostly barebones HTML document."
                        }
                     }
                  ]
               }
            }
         ]
      }
   }
]
```

Actual output:

```
[{"e":{"t":"main","a":{"role":"main","class":"container","id":"main_id"},"b":[{"e":{"t":"div","p":"main_id","a":{"class":"starter-template","id":"str_id"},"b":[{"e":{"t":"p","p":"str_id","a":{"class":"lead"},"b":"Use this document as a way to quickly start any new project.\n All you get is this text and a mostly barebones HTML document."}}]}}]}}]
```

bblanchon, 2022-05-08:
Hi @rjjrbatarao,

Thank you very much for reporting this issue.
The problem was in `doc[null]`, which clearly showed that the index was ignored.

I updated the assistant
Please try again and close this issue if the problem is fixed.

Best regards,
Benoit

rjjrbatarao, 2022-05-08:
Amazing I can now use my js library collagejs to build programmable frontend using arduinoJson, works perfectly Thank you very much.
https://github.com/rjjrbatarao/collagejs

closed 2022-05-08 by rjjrbatarao