bblanchon/ArduinoJson #1752: serializeJson?

whes1015, 2022-05-15:
After serializeJson, do I need to free memory? I have an infinitely increasing string (+=), after serializeJson I make the string (=""), it can keep running normally at first, the result after serializeJson after a while string becomes null

whes1015, 2022-05-15:
```c
DynamicJsonDocument json(8192);
char json_string[8192];
serializeJson(json, json_string);
```

bblanchon, 2022-05-16:
Hi @whes1015,

This sounds like a [fragmentation issue](https://cpp4arduino.com/2018/11/06/what-is-heap-fragmentation.html), but you didn't provide enough information.
Please follow one of the issue templates: https://github.com/bblanchon/ArduinoJson/issues/new/choose

Best regards,
Benoit

closed 2022-05-17 by whes1015