bblanchon/ArduinoJson #1755: fragmentation issue

whes1015, 2022-05-17:
**Describe the bug**  
fragmentation issue

**Troubleshooter report**  
The string will become `null` ,But it can display on Serial

**Environment**  
Here is the environment that I used:
* Microcontroller: ESP32
* Core/runtime: ESP32 1.0.6
* IDE: Arduino IDE 1.8.19

**Reproduction**  
```C++
#include <ArduinoJson.h>

DynamicJsonDocument json(4096);

void setup() {
  Serial.begin(115200);
}

void loop() {
  json["test"] = test;
  test += "test ";
  serializeJson(json, Serial);
}
```

**Compiler output**    
```
esptool.py v3.0-dev
Serial port COM3
Connecting........_____....._____.
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 8c:aa:b5:94:f0:ac
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 5957.9 kbit/s)...
Hash of data verified.
Compressed 17120 bytes to 11164...
Writing at 0x00001000... (100 %)
Wrote 17120 bytes (11164 compressed) at 0x00001000 in 0.1 seconds (effective 971.3 kbit/s)...
Hash of data verified.
Compressed 207008 bytes to 108074...
Writing at 0x00010000... (14 %)
Writing at 0x00014000... (28 %)
Writing at 0x00018000... (42 %)
Writing at 0x0001c000... (57 %)
Writing at 0x00020000... (71 %)
Writing at 0x00024000... (85 %)
Writing at 0x00028000... (100 %)
Wrote 207008 bytes (108074 compressed) at 0x00010000 in 1.5 seconds (effective 1074.0 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.0 seconds (effective 2048.0 kbit/s)...
Hash of data verified.
```

**Program output**  
`null`

Actual output:
`null`

whes1015, 2022-05-17:
I don't know if the string is too long, but Serial.print can display the string correctly

bblanchon, 2022-05-19:
Hi @whes1015,

Please read [I found a memory leak in the library!](https://arduinojson.org/v6/issues/memory-leak/).

Best regards,
Benoit

closed 2022-05-20 by whes1015