bblanchon/ArduinoJson #1757: Arducam and ArduinoJson library confliction

victorzuev8, 2022-05-17:
I'm using ArduCAM for an IoT project, and I need ArduinoJson library in my code. Unfortunately, when compiling basic blank code (setup and loop are empty). I know the problem is between ArduCam and ArduinoJSON basically I isolated the libraries and got the same error. I'm using Arduino IDE and MKR Wifi 1010. 

**My code**
#include <ArduCAM.h>
#include <ArduinoJson.h>

void setup() {
  
}

void loop() {
  
}

**Compiler output**    
Arduino: 1.8.12 (Windows 10), Board: "Arduino MKR WiFi 1010"

In file included from c:\users\babyb\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\exception:142:0,

                 from c:\users\babyb\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\ios:39,

                 from c:\users\babyb\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\ostream:38,

                 from C:\Users\babyb\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Strings/String.hpp:10,

                 from C:\Users\babyb\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Strings/Adapters/JsonString.hpp:9,

                 from C:\Users\babyb\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Strings/StringAdapters.hpp:8,

                 from C:\Users\babyb\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Memory/MemoryPool.hpp:10,

                 from C:\Users\babyb\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantData.hpp:7,

                 from C:\Users\babyb\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/SlotFunctions.hpp:8,

                 from C:\Users\babyb\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Array/ArrayIterator.hpp:7,

                 from C:\Users\babyb\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Array/ArrayRef.hpp:8,

                 from C:\Users\babyb\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:24,

                 from C:\Users\babyb\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from C:\Users\babyb\Desktop\test\test.ino:2:

c:\users\babyb\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\bits\exception_ptr.h:124:68: error: macro "swap" requires 3 arguments, but only 1 given

         exception_ptr(static_cast<exception_ptr&&>(__o)).swap(*this);

                                                                    ^

c:\users\babyb\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\bits\exception_ptr.h:132:26: error: macro "swap" requires 3 arguments, but only 1 given

       swap(exception_ptr&) _GLIBCXX_USE_NOEXCEPT;

                          ^

c:\users\babyb\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\bits\exception_ptr.h:166:52: error: macro "swap" requires 3 arguments, but only 2 given

     swap(exception_ptr& __lhs, exception_ptr& __rhs)

                                                    ^

c:\users\babyb\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\bits\exception_ptr.h:167:23: error: macro "swap" requires 3 arguments, but only 1 given

     { __lhs.swap(__rhs); }

                       ^

In file included from c:\users\babyb\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\bits\move.h:54:0,
[... 608 lines cut ...]

c:\users\babyb\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\istream:876:12: error: macro "swap" requires 3 arguments, but only 1 given

  swap(__rhs);

            ^

c:\users\babyb\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\istream:881:33: error: macro "swap" requires 3 arguments, but only 1 given

       swap(basic_iostream& __rhs)

                                 ^

c:\users\babyb\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\c++\7.2.1\istream:882:35: error: macro "swap" requires 3 arguments, but only 1 given

       { __istream_type::swap(__rhs); }

                                   ^

exit status 1
Error compiling for board Arduino MKR WiFi 1010.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

bblanchon, 2022-05-18:
Update: to work around this problem you can either define both `ARDUINOJSON_ENABLE_STD_STREAM` and `ARDUINOJSON_ENABLE_STD_STRING` to `0`.

```c++
#define ARDUINOJSON_ENABLE_STD_STREAM 0
#define ARDUINOJSON_ENABLE_STD_STRING 0
#include <ArduCAM.h>
#include <ArduinoJson.h>
```

or include `ArduCAM.h` after `ArduinoJson.h`:

```c++
#include <ArduinoJson.h>
#include <ArduCAM.h>
```

victorzuev8, 2022-05-18:
That worked, thank you very much.

closed 2022-05-18 by victorzuev8