bblanchon/ArduinoJson #1759: Clean and resue nested object

guestisp, 2022-05-19:
let's suppose this json and pseudocode:

```
DynamicJsonDocument docResponse(2048);
docResponse["jsonrpc"] = jsonrpc;
docResponse["id"] = requestId;
JsonObject result = docResponse.createNestedObject("result");
result["test"] = "hello world";

while ( x == y) {
   result["anothertest"] = "foo";
   String jsonResponse;
   serializeJson(docResponse, jsonResponse);
   replyToCustomer(docResponse);
}
```

it works. but now, after the while, I need to totally remove the result node and create a new one.
I've tried to set it as "" or overwriting with another `JsonObject result = docResponse.createNestedObject("result");` but I think there is some memory leak because after a while it crash (or stops to serialize the response)

How can I `.clear()` only the result node and not the full json document ?

bblanchon, 2022-05-21:
Hi @guestisp,

Indeed, the monotonic allocator in [`JsonDocument`](https://arduinojson.org/v6/api/jsondocument/) isn't able to release only a part of the document: it's all or nothing.
See [I found a memory leak in the library!](https://arduinojson.org/v6/issues/memory-leak/)

As a workaround, you can:

1. Call [`JsonDocument::clear()`](https://arduinojson.org/v6/api/jsondocument/clear/)  and re-add the three values
2. Call [`JsonDocument::garbageCollect()`](https://arduinojson.org/v6/api/jsondocument/garbagecollect/)
3. Use two `JsonDocument`s and call [`JsonVariant::link()`](https://arduinojson.org/v6/api/jsonvariant/link/)

If I were you, I would go with option 1 because it's the simplest and the impact on performance is neglectable.

See also [How to reuse a JsonDocument?](https://arduinojson.org/v6/how-to/reuse-a-json-document/)

Best regards,
Benoit

guestisp, 2022-05-24:
i did something similiar, i create a brand new document every time. i don't need a lot of performance, reliability is more important than performance here, so instead of creating a JSON at the start of the method, i create the JSON just before the point where i need the reply

bblanchon, 2022-05-29:
I'm glad you found a solution.
Can we close this issue?

guestisp, 2022-05-29:
sure.

closed 2022-05-29 by guestisp