bblanchon/ArduinoJson #1760: ESP8266 crash with 6.19.1+

waclaw66, 2022-05-20:
**Describe the bug**  
DynamicJsonDocument is crashing using 6.19.1+ version.
Last properly working version is 6.19.0.

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  
1. The issue happens at run time
2. The issue concerns serialization
3. Program crashes
4. Program uses <code>DynamicJsonDocument</code>
5. Program doesn’t use <code>PROGMEM</code>
6. Output type is <code>String</code>

**Environment**  
Here is the environment that I used:
* Microcontroller: ESP8266
* Core/runtime: ESP8266 with PlatformIO
* IDE: VSCode 1.67.1

**Reproduction**  
few loops of the following are enought to crash:
```c++
void test() {
    DynamicJsonDocument json(1024);
    JsonObject deviceJson = json.createNestedObject("device");
    deviceJson["identifiers"][0] = "meteo1";
    deviceJson["name"] = "thermometer";
    deviceJson["manufacturer"] = "me";
    deviceJson["model"] = "ESP8266";
    json["name"] = "Temperature";
    json["object_id"] = "meteo1";
    json["unique_id"] = "meteo1";
    json["device_class"] = "thermometer";
    json["unit_of_measurement"] = "*C";
    json["state_topic"] = "meteo/state";
    json["value_template"] = "{{ value_json.temperature }}";

    String s;
    serializeJson(json, s);
}

while (true) {
    test();
}
```

**Program output**  
Actual output:

```

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset

>>>stack>>>

ctx: sys
sp: 3ffff060 end: 3fffffb0 offset: 01a0
3ffff200:  3ffee2c0 3ffef670 3fff2050 00001f2b  
3ffff210:  00000000 00729cf6 4020f0d8 00000001  
3ffff220:  3ffef670 00000001 00000000 4020f0e3  
3ffff230:  00000001 000009f9 3fff208c 402096a4  
3ffff240:  00000000 0000013c 0000000d 00000001  
3ffff250:  00000000 00000000 401002c6 3fff7733  
```

bblanchon, 2022-05-20:
Hi @waclaw66,

The exception `Soft WDT reset` is expected when you hold the execution for too long.
This is not specific to any ArduinoJson version; in fact, it's not specific to ArduinoJson.
You can reproduce this issue without the library; for example:

```c++
while (true) {}
```

To fix this issue, you must either:

1. call `yield()` from the loop
2. call `test()` from `loop()`

Best regards,
Benoit

waclaw66, 2022-05-20:
Hi @bblanchon,
I just wanted to demonstrate the loop itself with `while(true)`, loop takes <1s. I've updated the example.
It crashes in many ways, this time with the following....
```

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3584, room 16
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
```

The cause is in ArduinoJson >=6.19.1, previous versions run flawlessly.

bblanchon, 2022-05-21:
Please don't edit your question like this: it's very rude because it makes my answer look ill-suited.
The new version of your program runs flawlessly on my beloved Adafruit Feather HUZZAH.

waclaw66, 2022-05-21:
> Please don't edit your question like this: it's very rude because it makes my answer look ill-suited. The new version of your program runs flawlessly on my beloved Adafruit Feather HUZZAH.

Sorry for that. I'll try to find out which commit between 6.19.0 and 6.19.1 is causing the crashes.

waclaw66, 2022-05-23:
https://github.com/bblanchon/ArduinoJson/commit/4f6244eef415f2f3eb853243522434fb4a847df0 is causing that crash.

bblanchon, 2022-05-23:
Again, the sample code you shared doesn't crash my ESP8266 with ArduinoJson 6.19.4.

waclaw66, 2022-05-25:
Here is the crashing example. Probably somehow related to mysensors. Nevertheless 6.19.0 works fine.

```c++
#define MY_DEBUG
#define MY_BAUD_RATE 115200
#define MY_GATEWAY_ESP8266
#define MY_GATEWAY_MQTT_CLIENT
#define MY_CONTROLLER_URL_ADDRESS "mqtt.example.com"
#define MY_PORT 1883
#define MQTT_MAX_PACKET_SIZE 1024
#define MQTT_KEEPALIVE 60
#define MQTT_SOCKET_TIMEOUT 60
#define MY_MQTT_USER "mqtt"
#define MY_MQTT_PASSWORD "password"
#define MY_WIFI_SSID "ssid"
#define MY_WIFI_PASSWORD "password"
#define MY_MQTT_CLIENT_ID "meteo"

#include <MySensors.h>
#include <ArduinoJson.h>

PubSubClient mqtt = _MQTT_client;
unsigned long publishInterval = 2 * 1000;
unsigned long lastPublishTime = 0;

void publish(String topic, const char *json)
{
#ifdef MY_DEBUG
    Serial.println(topic + ": (" + String(strlen(json)) + " B) " + json);
#endif
    if (!mqtt.publish(topic.c_str(), json)) {
        Serial.println("Send failed!");
    }
}

void publish(String topic, JsonDocument &json)
{
    String s;
    serializeJson(json, s);
    publish(topic, s.c_str());
}

void discovery(String identifier, String name, String desc, const char *cls = NULL, const char *unit = NULL)
{
    DynamicJsonDocument json(1024);
    JsonObject deviceJson = json.createNestedObject("device");
    deviceJson["identifiers"][0] = identifier;
    deviceJson["name"] = identifier;
    deviceJson["manufacturer"] = "me";
    deviceJson["model"] = "ESP8266";
    json["name"] = desc;
    String id = identifier + "_" + name;
    json["object_id"] = id;
    json["unique_id"] = id;
    if (cls) {
        json["device_class"] = cls;
    }
    if (unit) {
        json["unit_of_measurement"] = unit;
    }
    json["state_topic"] = identifier + "/state";
    json["value_template"] = "{{ value_json." + name + " }}";

    publish(String("homeassistant/sensor/") + id + "/config", json);
}

void presentation()
{
    if (!mqtt.connected()) return;
    discovery(MY_MQTT_CLIENT_ID, "ip", "IP address");
    discovery(MY_MQTT_CLIENT_ID, "temperature", "Air temperature", "temperature", "°C");
    discovery(MY_MQTT_CLIENT_ID, "pressure", "Air pressure", "pressure", "hPa");
    discovery(MY_MQTT_CLIENT_ID, "humidity", "Air humidity", "humidity", "%");
}

void setup()
{
}

void loop()
{
    if (!mqtt.connected()) return;
    unsigned long now = millis();
    if (lastPublishTime && (unsigned long)(now - lastPublishTime) < publishInterval) return;
    lastPublishTime = now;
    presentation();
    DynamicJsonDocument json(256);
    json["ip"] = WiFi.localIP();
    json["temperature"] = 20.0;
    json["pressure"] = 1000.0;
    json["humidity"] = 70.0;
    publish(String(MY_MQTT_CLIENT_ID) + "/state", json);
}
```

platformio.ini
```ini
[env:nodemcuv2]
platform = espressif8266@2.6.3
board = nodemcuv2
framework = arduino
monitor_speed = 115200
build_type = debug
lib_deps =
	mysensors/MySensors@^2.3.2
	bblanchon/ArduinoJson@6.19.4
```

bblanchon, 2022-05-27:
Hi @waclaw66,

I don't see any obvious problem, except for the abusive use of `String`.
Can you try to upgrade the platform?

Also, can you confirm that the program in your original message didn't crash the ESP8266?

Best regards,
Benoit

waclaw66, 2022-05-27:
Hi @bblanchon ,

> I don't see any obvious problem, except for the abusive use of `String`. Can you try to upgrade the platform?

Unfortunatelly I can't, 2.6.3 is the last properly working version, version 3.x.x freezes at start.

 
> Also, can you confirm that the program in your original message didn't crash the ESP8266?

Yes, the first example didn't crash, only while using it with MySensors.

bblanchon, 2022-05-28:
Honestly, I don't think ArduinoJson is to blame, so I'll close this issue.
To me, there is an undefined behavior in your program that accidentally works with 6.19.0.
Please reopen a new issue as soon as you can provide a reproduction code that only involves ArduinoJson.

Now, I recommend you start by figuring out why the newer versions of the framework don't work.
First, make sure you selected the right board.
Then, try with another (preferable brand name) board.
If it's still not working, please open an issue on [the dedicated GitHub project](https://github.com/platformio/platform-espressif8266).

closed 2022-05-28 by bblanchon