bblanchon/ArduinoJson #1762: Error on sending data with ArduinoJSON

adriankrz1, 2022-05-29:
**Description**
When I upload the code with Json data, and when I serialize it inside the loop, I can only send the data to the database once, and then it just displays me the code that something is not correct, (http error code 400). If I put these data on the void setup(), the code works fine, but when on loop, it only works the first time. Also I checked and there is no overflow. I also tried the dynamicJson, but the problem was the same

**Troubleshooter's report**
1. The issue happens at run time
2. The issue concerns serialization
3. Output is incomplete
4. Program calls <code>serializeJson()</code> in a loop
5. Moving the <code>JsonDocument</code> inside the loop doesn’t solve the issue
6. <code>JsonDocument::overflowed()</code> returns <code>false</code>
7. No string contains a NUL

**Environment**
* Microcontroller: ESP8266
* Core/Framework: ESP8266 core for Arduino v2.7.4
* IDE: Arduino IDE 1.8.13

**Reproduction code**
```c++
 StaticJsonDocument<192> doc;
 doc["PMCO"] = "1.1";
    doc["PMNO2"] = "1.2";
    doc["CO2"] = "1.3";
    doc["PM25"] = "1.4";
    doc["PM10"] = "1.5";
    doc["podid"] = "10";
serializeJson(doc, httpRequestData);
```

bblanchon, 2022-05-30:
Hi @adriankrz1,

You told the Assistant that the "Output is incomplete"; can you print the output?

Best regards,
Benoit

adriankrz1, 2022-05-30:
Hello,
The output seems okay, the problem I realized is that the program outputs
keep adding. So the first time the output is okay, the second time, it just
adds the first output and the second. The problem is that I can't clear the
first output, and when on the loop, the output keeps adding up.
Check the photo attached. I just need to know how to clear the last json so
that every time on the loop, the output is only the last one.

Thank you,
Adrian Kryeziu

bblanchon, 2022-05-31:
Hi Adrian,

Indeed, [serializeJson()](https://arduinojson.org/v6/api/json/serializejson/) appends to `String`.
To fix your issue, you must either move the `String` declaration in the loop (so that a new instance is created on each iteration) or reset the content of the `String` before calling `serializeJson()`.
Alternatively, you can pass the TCP stream directly to `serializeJson()`.

Best regards,
Benoit

closed 2022-06-10 by bblanchon