bblanchon/ArduinoJson #1765: Is binary blob allowed with msgpack?

hyansuper, 2022-06-02:
I know json does not allow binary. but I am using msgpack, can I serialize/deserialze a blob of binary data using msgpack?
Supporse I want to transfer a picture, but when the binary data contains 0, The output is truncated. 
```
#include <ArduinoJson.h>

void setup() {
  Serial.begin(115200);    
  uint8_t raw[]={0,101,102,103,104,105,106,107,108,109};
  StaticJsonDocument<512> doc;  
  doc["fmt"]="jpeg";
  doc["data"] = raw;
  Serial.println("serialize:");
  serializeMsgPack(doc, Serial);
  // output: ⸮⸮fmt⸮jpeg⸮data⸮
}

void loop() {
}
```
Or should I use [JsonArray](https://arduinojson.org/v6/api/jsonarray/)? can I initiate it with raw data, something like `JsonArray(buf=raw)` ?

hyansuper, 2022-06-04:
thank you! I use msg pack because I need a RPC that supports binary data streaming as well as other data types. 
using ArduinoJson is really handy, and the non-duplication feature makes it efficient

hyansuper, 2022-06-04:
Oh, some related questions:
1. `doc["data"] = JsonString((const char*)raw, sizeof(raw));` sends binary blob as string, but the receiving device does not use ArduinoJson, will it be interpreted as string when deserialize? will additional NUL characters be inserted into the raw data?
2. I don't know how raw data and strings are coded into msgpack format respectivly (I guess raw data is send as is with some additional overhead, but strings are compressed/coded), if I treat raw data as strings in the method above, will it be conpressed? cause the compression process may slow down transfer speed, considering efficiency.

bblanchon, 2022-06-05:
1. Make sure the receiver doesn't try to decode UTF-8 characters.
2. There is no translation, nor compression in the MessagePack format.

closed 2022-07-05 by bblanchon