bblanchon/ArduinoJson #1767: Filtering/Chunks based on key value

guestisp, 2022-06-10:
Let's assume a very big document (1MB or more). It's a temperature log:
```
{
	"2022-01-01": {
		"00:00:00": {"ph": 12.34, "temp": 12.34, "status": 1, "ml": 12.34},
		"01:00:00": {"ph": 12.34, "temp": 12.34, "status": 1, "ml": 12.34},
		"02:00:00": {"ph": 12.34, "temp": 12.34, "status": 1, "ml": 12.34},
		"03:00:00": {"ph": 12.34, "temp": 12.34, "status": 1, "ml": 12.34},
		"04:00:00": {"ph": 12.34, "temp": 12.34, "status": 1, "ml": 12.34},
		"05:00:00": {"ph": 12.34, "temp": 12.34, "status": 1, "ml": 12.34},
		"06:00:00": {"ph": 12.34, "temp": 12.34, "status": 1, "ml": 12.34},
		"07:00:00": {"ph": 12.34, "temp": 12.34, "status": 1, "ml": 12.34},
		"08:00:00": {"ph": 12.34, "temp": 12.34, "status": 1, "ml": 12.34},
		"09:00:00": {"ph": 12.34, "temp": 12.34, "status": 1, "ml": 12.34},
                ...............................
	"2022-01-02": {
		"00:00:00": {"ph": 12.34, "temp": 12.34, "status": 1, "ml": 12.34},
		"01:00:00": {"ph": 12.34, "temp": 12.34, "status": 1, "ml": 12.34},
		"02:00:00": {"ph": 12.34, "temp": 12.34, "status": 1, "ml": 12.34},
		"03:00:00": {"ph": 12.34, "temp": 12.34, "status": 1, "ml": 12.34},
		"04:00:00": {"ph": 12.34, "temp": 12.34, "status": 1, "ml": 12.34},
		"05:00:00": {"ph": 12.34, "temp": 12.34, "status": 1, "ml": 12.34},
		"06:00:00": {"ph": 12.34, "temp": 12.34, "status": 1, "ml": 12.34},
		"07:00:00": {"ph": 12.34, "temp": 12.34, "status": 1, "ml": 12.34},
		"08:00:00": {"ph": 12.34, "temp": 12.34, "status": 1, "ml": 12.34},
		"09:00:00": {"ph": 12.34, "temp": 12.34, "status": 1, "ml": 12.34},
```

and so on. One object for every day containing an object with each probe time (hourly) and the readings.

as it would be huge, I would like to set an upper limit, like 365, so when I write the today reading, before appending the new value, I would like to remove all entries regarding today - 365 days ago (the "expired" logs)

Any idea ? I don't think I can use filtering or chunked deserialize.

I can modify the json, but only the hourly properties, in example, by removing the key and using an array:
```
"2022-01-02": [
                {"hour": "00:00:00", "ph": 12.34, "temp": 12.34, "status": 1, "ml": 12.34},
		{"hour": "01:00:00", "ph": 12.34, "temp": 12.34, "status": 1, "ml": 12.34},
```

bblanchon, 2022-06-13:
Hi @guestisp,

I recommend using the [JSONL format](https://jsonlines.org/) to log information with ArduinoJson.
In your case, a file would look like so:

```jsonl
{"timestamp":"2022-01-01T00:00:00","ph": 12.34, "temp": 12.34, "status": 1, "ml": 12.34}
{"timestamp":"2022-01-01T01:00:00","ph": 12.34, "temp": 12.34, "status": 1, "ml": 12.34}
{"timestamp":"2022-01-01T02:00:00","ph": 12.34, "temp": 12.34, "status": 1, "ml": 12.34}
```

The advantage of this technique is that you don't need to read the file to append new data.

You can easily rotate the files by using a different filename for each day.

See also:

* [How to append a JSON object to a file?](https://arduinojson.org/v6/how-to/append-to-a-file/)

Best regards,
Benoit

guestisp, 2022-06-13:
that's perfect. thank you

closed 2022-06-14 by bblanchon