bblanchon/ArduinoJson #1768: Serialize JSON sometimes causes system crash while writing to SD stream (ESP8266)

xpalaa00, 2022-06-12:
**Issue description**  
Hello there. In my project I am trying to grab data from sensors which are sequently pushed into a DynamicJsonDocument (it is large JSON document ... allocated 1536). After that I am trying to serialize these data into my SD card via SDFS stream (due to other issues the SPI driving frequency is needed to be underclocked... 200 kHz). Usually everything is done without issues, but sometimes saving process take a little more time (longer than 2 seconds) and then the software watchdog cause system restart (Wi-Fi connection should be closed during writing process). I tried to use yield() function before using serializeJsonPretty(), but it doesn't take any effect. Disabling of software watchdog isn't a way, because it will trigger a hardware watchdog instead. I am not sure, how to solve this issue. Is there any way? Thanks for helping hand in advance.

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  
1. The issue happens at run time
2. The issue concerns serialization
3. Program crashes
4. Program uses <code>DynamicJsonDocument</code>
5. Program doesn’t use <code>PROGMEM</code>
6. Output type is <code>Stream</code>

**Environment**  
Here is the environment that I'm using':
* Microconroller: ESP8266 (NodeMCU V3)
* Core/runtime: ESP8266 core for Arduino v3.0.2
* IDE: MicroSoft Visual Studio (v17.2.3) with Visual Micro module (v2022.607.2200)

**Stack output after crash**  
Decoding stack results
0x40215d50: SPIClass::transferBytes_(unsigned char const*, unsigned char*, unsigned char) at ...\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\SPI\SPI.cpp line 602
0x40100519: millis() at ...\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_wiring.cpp line 193
0x40215ddc: SPIClass::transferBytes(unsigned char const*, unsigned char*, unsigned int) at ...\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\SPI\SPI.cpp line 539
0x40218c36: sdfat::SdSpiArduinoDriver::send(unsigned char const*, unsigned int) at ...\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266SdFat\src\SpiDriver\SdSpiESP.cpp line 86
0x4021883d: sdfat::SdSpiCard::writeData(unsigned char, unsigned char const*) at ...\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266SdFat\src\SdCard/SdSpiCard.h line 327
0x402188a5: sdfat::SdSpiCard::writeData(unsigned char const*) at ...\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266SdFat\src\SdCard\SdSpiCard.cpp line 718
0x402189cc: sdfat::SdSpiCard::writeSectors(unsigned int, unsigned char const*, unsigned int) at ...\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266SdFat\src\SdCard\SdSpiCard.cpp line 801
0x40215fec: sdfat::FatFile::addDirCluster() at ...\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266SdFat\src/SdCard/SdSpiCard.h line 229
0x4021627d: sdfat::FatFile::readDirCache(bool) at ...\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266SdFat\src\FatLib\FatFile.cpp line 878
0x402175a7: sdfat::FatFile::open(sdfat::FatFile*, sdfat::fname_t*, unsigned char) at ...\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266SdFat\src\FatLib\FatFileLFN.cpp line 445
0x402168bc: sdfat::FatFile::open(sdfat::FatFile*, char const*, unsigned char) at ...\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266SdFat\src\FatLib\FatFile.cpp line 469
0x40101206: umm_free(void*) at ...\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 595
0x40101224: umm_poison_free_fl(void*, char const*, int) at ...\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\umm_malloc/umm_local.c line 152
0x402168ec: sdfat::FatFile::open(sdfat::FatVolume*, char const*, unsigned char) at ...\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266SdFat\src\FatLib/FatVolume.h line 340
0x40215641: sdfs::SDFSImpl::open(char const*, fs::OpenMode, fs::AccessMode) at ...\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266SdFat\src/FatLib/FatFile.h line 389
0x40227c85: vsnprintf at /workdir/repo/newlib/newlib/libc/stdio/vsnprintf.c line 40
0x4021d34c: Print::printf(char const*, ...) at ...\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\Print.cpp line 66
0x4021cbdc: fs::FS::open(char const*, char const*) at ...\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed\xtensa-lx106-elf\include\c++\10.3.0\bits/shared_ptr_base.h line 1324
0x4020c350: MeasureManager::saveDataToJson_(fs::FS*, unsigned int) at ...\AppData\Local\Temp\VMBuilds\IoT-Meteostanice\esp8266_nodemcuv2\Debug\measurement\MeasureManager.cpp line 1728
0x4021b954: optimistic_yield(uint32_t) at ...\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 153
0x4021e48c: uart_write(uart_t*, char const*, size_t) at ...\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\uart.cpp line 546
0x4021cdbc: HardwareSerial::write(unsigned char const*, unsigned int) at ...\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/HardwareSerial.h line 193
0x4021d280: Print::write(char const*) at ...\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/Print.h line 59
0x4020ea70: MeasureManager::synchronizeData_() at ...\AppData\Local\Temp\VMBuilds\IoT-Meteostanice\esp8266_nodemcuv2\Debug\measurement\MeasureManager.cpp line 3036
0x4020eb7e: MeasureManager::loop() at ...\AppData\Local\Temp\VMBuilds\IoT-Meteostanice\esp8266_nodemcuv2\Debug\measurement\MeasureManager.cpp line 322
0x402062b8: loop() (...\Desktop\IoT-Meteostanice at nová)\src/IoT-Meteostanice.ino line 120
0x4021b99c: loop_wrapper() at ...\AppData\Local\arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 201

bblanchon, 2022-06-13:
Hi @xpalaa00,

```c++
SPIClass::transferBytes_(unsigned char const*, unsigned char*, unsigned char)
millis()
SPIClass::transferBytes(unsigned char const*, unsigned char*, unsigned int)
sdfat::SdSpiArduinoDriver::send(unsigned char const*, unsigned int)
sdfat::SdSpiCard::writeData(unsigned char, unsigned char const*)
sdfat::SdSpiCard::writeData(unsigned char const*)
sdfat::SdSpiCard::writeSectors(unsigned int, unsigned char const*, unsigned int)
sdfat::FatFile::addDirCluster()
sdfat::FatFile::readDirCache(bool)
sdfat::FatFile::open(sdfat::FatFile*, sdfat::fname_t*, unsigned char)
sdfat::FatFile::open(sdfat::FatFile*, char const*, unsigned char)
umm_free(void*)
umm_poison_free_fl(void*, char const*, int)
sdfat::FatFile::open(sdfat::FatVolume*, char const*, unsigned char)
sdfs::SDFSImpl::open(char const*, fs::OpenMode, fs::AccessMode)
vsnprintf
Print::printf(char const*, ...)
fs::FS::open(char const*, char const*)
MeasureManager::saveDataToJson_(fs::FS*, unsigned int)
optimistic_yield(uint32_t)
uart_write(uart_t*, char const*, size_t)
HardwareSerial::write(unsigned char const*, unsigned int)
Print::write(char const*)
MeasureManager::synchronizeData_()
MeasureManager::loop()
loop()
loop_wrapper()
```

ArduinoJson isn't in this stack trace, why do put the blame on it?

Best regards,
Benoit

xpalaa00, 2022-06-13:
Ouh... I apologize very much. It's the thirst time I have used the ESP exception decoder tool and I wasn't sure, what that informations in the stack meant. Is it right, stack contains pointers to last called sequence of functions/methods from deepest called to upper and is used by return functions, or is there another sort order?

But why I thought the ArduinoJson is cause of issue? I'm not sure right now, because I lost the log file containing this failure, but if I remember it right, in my serial monitor log I found message "file opened" which is called after success file open method (success is determined by if statement "if (file) { ... }") and next called function is function jsonSerializePretty(...).

However, it would be better to rely on the stack content. So, if I pass through the stack, there are some data writing methods propably called inside file open method that cause the crash, right? Than I want to apologize again, the ArduinoJson is powerfull and very usefull library and this issue is not related to ArduinoJson library, I am sorry, but every time I see this issue in my console randomly occuring, I feel a little bit hopeless. It isn't neccesary to ansfer this, because it isn't ArduinoJson related issue, but do You have any ideas, what can cause it exactly, how to solve it, or just where I should continue?

Thank You heartly.

closed 2022-06-13 by xpalaa00

bblanchon, 2022-06-14:
Since the issue comes from the SD library on ESP8266, you should ask [the authors](https://github.com/esp8266/Arduino/) for help.
Before contacting them, though, I recommend you perform the following checks:

1. Test the same program with version 2.7.4 of the Arduino core for ESP8266
2. Test with another SD card
3. Write to a temporary buffer (for example, a `String`) before writing to the file (instead of passing the file to `serializeJson()`)
4. Make sure you call `File::close()`

xpalaa00, 2022-06-14:
Thank You for Yours ideas. I will continue to looking around for solution. For now, I have increased a little bit a speed of SPI bus in range of possibility, so reading, writing and opening time is shorter and the crash didn't occured yet, but I think it is still possible it will take a chance to occure. And again I am sorry for my wrong analysis of the problem cause.