bblanchon/ArduinoJson #1769: no default constructor exists for class "ArduinoJson6194_F1::BasicJsonDocument<ArduinoJson6194_F1::DefaultAllocator>"

JacobMaldonado, 2022-06-14:
**Describe the issue**  
A clear and concise description of what you're trying to do.
You don't need to explain every aspect of your project: focus on the problem you're having.

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  
[Paste the report here]

**Environment**  
Here is the environment that I'm using':
* Microconroller: [e.g. ESP8266]
* Core/runtime: [e.g. ESP8266 core for Arduino v3.0.2]
* IDE: [e.g. Arduino IDE 1.8.16]

**Reproduction**  
Here is a small snippet that demonstrate the problem.

```c++
DynamicJsonDocument doc(1024);

DeserializationError error = deserializeJson(doc, "{\"hello\":\"world\"}");

// insert code here
```

**Program output**  
If relevant, include the program output.

Expected output:

```
[insert expected output here]
```

Actual output:

```
[insert actual output here]
```

bblanchon, 2022-06-16:
Hi @JacobMaldonado,

This is the same error as [no matching function for call to 'BasicJsonDocument::BasicJsonDocument()'](https://arduinojson.org/v6/error/no-matching-function-for-call-to-basicjsondocument-basicjsondocument/).

To fix it, you must pass the memory pool capacity to [`DynamicJsonDocument`](https://arduinojson.org/v6/api/dynamicjsondocument/)'s constructor, like so:

```c++
class Configuration {
public:
  Configuration() : _config(1024) {}

private:
  DynamicJsonDocument _doc;
};
```

Also, remove the destructor as it can only crash your program.

Best regards,
Benoit

JacobMaldonado, 2022-06-21:
@bblanchon Thank you so much.

closed 2022-06-21 by JacobMaldonado