bblanchon/ArduinoJson #1770: Returning a pointer (or null) when searching for a JsonObject by key?

burtonkent, 2022-06-16:
**Troubleshooter report**  
(My issue isn't listed)

**Environment**  
Here is the environment that I'm using':
* ESP32
* Core/runtime: Expressif for ESP32 on Arduino / platformio/framework-arduinoespressif32
* IDE: platform.io with Visual Studio Code

**Issue**
I'm trying to make a function that searches for a key even if nested in the JSON document. It's based on [the first "trick" on this page ](https://arduinojson.org/v5/doc/tricks/)(which is for ArduinoJson 5 not 6). Note the errors depending on how I'm returning a reference. I'm returning a pointer so I can return NULL if not found.

    JsonObject* findNestedObject(const JsonObject &obj, const String &key) {
        for (const JsonPair pair: obj) {
            if (pair.key().c_str() == key.c_str()) {
                // error: taking address of temporary [-fpermissive]
                return &(pair.value().as<JsonObject>());

                // warning: address of local variable 'foundObject' returned
                JsonObject foundObject = pair.value().as<JsonObject>();
                return &foundObject;
            }

            if (pair.value().is<JsonObject>()) {
                JsonObject* nestedObject = this->findNestedObject(pair.value(), key);
                if (nestedObject) {
                    return nestedObject;
                }
            }     
        }
        return NULL;
    }

What's the correct way to return the needed JsonObject reference?

bblanchon, 2022-06-16:
Hi @burtonkent,

This is one of the fundamental differences between v5 and v6: [JsonObject](https://arduinojson.org/v6/api/jsonobject/) is now a reference (a smart pointer if you want) that points to the data in the memory pool.

Therefore, in ArduinoJson 6, you should never handle a `JsonObject` by address or by reference.
A NULL reference can be constructed with `JsonObject()`.

Here is one possible implementation:

```c++
JsonObject findNestedObject(JsonObjectConst obj, const String &key) {
  JsonObject foundObject = obj[key];
  if (foundObject) return foundObject;

  for (const JsonPair pair : obj) {
    JsonObject nestedObject = findNestedObject(pair.value(), key);
    if (nestedObject) return nestedObject;
  }

  return JsonObject();
}
```

Best regards,
Benoit

bblanchon, 2022-06-17:
Here it is: https://arduinojson.org/v6/how-to/search-nested-key/

bblanchon, 2022-06-17:
BTW, thank you very much for purchasing my book; your support means a lot to me!

closed 2022-06-17 by bblanchon