bblanchon/ArduinoJson #1773: Passing JsonDocument class member type to function to fill in JSON elements fails

milanbx, 2022-06-26:
Hi,
I am trying to pass class member type argument of JsonDocument to a simple function, where I want to modify the Json document.

**Issue description**
The class definition is following:
``` c++
class ESPConfig {
  public:
    using JsonDocument = StaticJsonDocument<JSON_DOC_SIZE>;
  ...
}
```
The class member function using the type is following:
``` c++
void ESPConfig::SetConfig() {
  JsonDocument jsonCfg;
  CurrentConfig2JSON(jsonCfg);
  ...
}
```
And the function called from the class member function is following:
```  c++
void CurrentConfig2JSON(ESPConfig::JsonDocument &json2Set) {
  json2Set['name'] = 'name';
  ...
}
```
And the single line of code above causes an exception and the board resets, when the function is called as following:
The reset cause reports this:
```
 ets Jan  8 2013,rst cause:2, boot mode:(1,7)
```

**Troubleshooter report**
Can not be applied.

**Environment**
Here is the environment that I'm using':
* Microconroller: [e.g. ESP8266]
* Core/runtime: [e.g. ESP8266 core for Arduino v3.0.2]
* IDE: VS Code + Arduino extension v0.4.12

Thanks for help.

milanbx, 2022-06-26:
Oh, I see it myself now, so clear:
``` c++
  json2Set["name"] = "name";
```
So stupid. Sorry, closing...

closed 2022-06-26 by milanbx

bblanchon, 2022-06-27:
Hi @milanbx,

C++ contains many pitfalls, so don't overlook the compiler warnings.
In your case, they should say:

> warning: character constant too long for its type
> warning: multi-character character constant [-Wmultichar]

Best regards,
Benoit

milanbx, 2022-06-27:
You are absolutely right, just to read the compiler warnings properly would save my time and your as well, as well as many other developers :-)
Thank you.
Rgds,
Milan