bblanchon/ArduinoJson #1778: how to caculate a String length within unicode char on use DynamicJsonDocument ?

brainsky, 2022-07-19:
**Description**
When I use DynamicJsonDocument  deserializeJson a string , I caculate it len  use String.length(), But I serializeJson to Serial, I found some key and value is lose.

**Troubleshooter's report**
1. The issue happens at run time
2. The issue concerns deserialization
3. <code>deserializeJson()</code> returns <code>Ok</code>
4. Some strings are truncated
5. The input contains <code>\u0000</code>
6. Using <code>JsonString</code> doesn’t solves the issue

**Environment**
* Microcontroller: ESP32
* Core/Framework: arduino esp32 idf-release/v4.4
* IDE: visual studio code 1.62.1

**Reproduction code**
```c++
String responseJson = "{\"data\":[{\"alarmTriggerSecond\":0,\"allDay\":false,\"attendeeNames\":[\"first\"],\"createTime\":\"2022-06-28T18:29:10\",\"description\":\"test query detail from file! else test deserializa json\",\"eventEndDate\":\"2022-06-28 18:42:00\",\"eventStartDate\":\"2022-06-28 18:12:00\",\"id\":\"62bad7f6c8c0970a3d1aa45d\",\"location\":\"beijing\",\"orgName\":\"ibu0001\",\"serverCalendarName\":\"默认日历\",\"serverInfoId\":43,\"subject\":\"test\",\"type\":4}],\"errCode\":0,\"id\":\"testId123456\",\"uri\":\"calendarevent\"}";
	
Serial.printf("responseJson len %d \r\n",  responseJson.length());

DynamicJsonDocument doc1(responseJson.length()) ;

deserializeJson(doc1, responseJson);

JsonObject obj = doc1.as<JsonObject>();

JsonObject subject  = obj["data"][0].as<JsonObject>();

serializeJson(subject, Serial);

//Serial print below:
responseJson len 466 
{"alarmTriggerSecond":0,"allDay":false,"attendeeNames":["first"],"createTime":"2022-06-28T18:29:10","description":"test query detail from file! else test deserializa json","eventEndDate":"2022-06-28 18:42:00","eventStartDate":"2022-06-28 18:12:00","id":"62bad7f6c8c0970a3d1aa45d","location":"beijing"}

```

**Remarks**
As we see Serial print, it lose some key and value, like orgName、subject。when I change DynamicJsonDocument  doc1 length big, and it is ok! I set DynamicJsonDocument doc1(1024);
//Serial print below:
{"alarmTriggerSecond":0,"allDay":false,"attendeeNames":["first"],"createTime":"2022-06-28T18:29:10","description":"test query detail from file! else test deserializa json","eventEndDate":"2022-06-28 18:42:00","eventStartDate":"2022-06-28 18:12:00","id":"62bad7f6c8c0970a3d1aa45d","location":"beijing","orgName":"ibu0001","serverCalendarName":"榛樿鏃ュ巻","serverInfoId":43,"subject":"test","type":4}

bblanchon, 2022-07-21:
Hi @brainsky,

Please see: 

* [Deserialization tutorial](https://arduinojson.org/v6/doc/deserialization/)
* [How to determine the capacity of the JsonDocument?](https://arduinojson.org/v6/how-to/determine-the-capacity-of-the-jsondocument/)

Best regards,
Benoit

brainsky, 2022-07-23:
> Hi @brainsky,
> 
> Please see:
> 
> * [Deserialization tutorial](https://arduinojson.org/v6/doc/deserialization/)
> * [How to determine the capacity of the JsonDocument?](https://arduinojson.org/v6/how-to/determine-the-capacity-of-the-jsondocument/)
> 
> Best regards, Benoit

Thank you so much for your reply, I really didn't find this article carefully before: How to determine the capacity of the JsonDocument? I'll try it out and see the results, and it should help me.

closed 2022-07-23 by brainsky