bblanchon/ArduinoJson #1779: deserializeJson return "ok" even in the wrong json format

PC-maniak, 2022-07-19:
Hello, I use your library, which is absolutely perfect =) by chance, I discovered the inconvenience of checking the correct processing of a json document

- Json dokument = {"T":"52"} -> OK everything works as it should

- Broken dokument = "T":"52"} -> the status "OK" is returned, but of course no value can be loaded
- if the last "}" is missing, "error" is returned

piece of code that checks correctness
```
// Deserialize the JSON document
      DeserializationError error = deserializeJson(root, buffer_char);
if (error) {
        Serial.println("BAD!");
        Serial.print(F("deserializeJson() failed: "));
        Serial.println(error.f_str());
      }
      else
      { 
 Serial.println("OK!");
}
```
Arduino IDE 1.8.13
Board "Arduino Micro pro"
ArduinoJson - 6.19.4

bblanchon, 2022-07-21:
Hi @PC-maniak,

`deserializeJson()` stops reading as soon as the JSON document ends.
In your case, it reads `"T"` and returns `Ok` immediately, ignoring the remaining part.
If you try to serialize the result, you see that it does contain `"T"`.

This feature enables [deserialization in chunks](https://arduinojson.org/v6/how-to/deserialize-a-very-large-document/#deserialization-in-chunks) and allows non-null-terminated input strings.

I added this scenario to the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/).
Let me know what you think.

Best regards,
Benoit

PC-maniak, 2022-07-21:
Aha! I didn't know that =) thanks for the info

closed 2022-07-22 by bblanchon