bblanchon/ArduinoJson #1780: Array of object AruinoV6

HDenied, 2022-07-25:
Hi,

I am trying to generate an array of object, I tried to follow this example in the doc but I am trying to write the object in a file with no much success:
    StaticJsonDocument<200> doc;
    JsonArray array = doc.to<JsonArray>();
    JsonObject nested = array.createNestedObject(); 
    nested["test"]="sprint";
    ofstream out_f {"my.json", ios::out};
    serializeJsonPretty(nested, out_f);

WHat I see printed is:

{"test":"sprint"}

What I expected is:

[{"test":"sprint"}]

Any idea of the problem? Am I missing something?

bblanchon, 2022-07-26:
Hi @HDenied,

You must replace `serializeJsonPretty(nested, out_f)` with `serializeJsonPretty(doc, out_f)`.

Best regards,
Benoit

HDenied, 2022-07-26:
I figured it out, thanks, btw, I really like your library and the style you wrote it, there is much to learn from your code

bblanchon, 2022-07-27:
You're welcome!
Thank you for using ArduinoJson.

closed 2022-07-27 by bblanchon