bblanchon/ArduinoJson #1782: Don't know how big to dimension the JSON Doc

basementmedia2, 2022-07-30:
**Describe the issue**  
Hi,

short & concise question:
I have a function that scans for networks and writes them to a JSON document.

Theoretically there could be a billion networks in the environment.
So I don't know how big I have to dimension "networks_arr" (please see Sketch below).

So here is the question:
What would happen in the worst case, if the found networks do not all "fit in"? 
Does then the memory run full and the ESP says goodbye?
Or are then simply not all found included in the JSON Doc?

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  
[Paste the report here]

**Environment**  
Here is the environment that I'm using':
ESP32 Wemos D1 

**Reproduction**  
Here is a small snippet that demonstrate the problem.

```
void scan_networks() {
  StaticJsonDocument<1536> networks_arr; // Hier kann ich die Grüße nur schätzen
  int n = WiFi.scanNetworks();
  if (n == 0) {
      Serial.println("Keine Netzwerke gefunden");
  } else {
    for (int i = 0; i < n; ++i) {
        networks_arr[i]["net_ssid"] = WiFi.SSID(i);
        networks_arr[i]["net_rssi"] = WiFi.RSSI(i);
    }
    serializeJson(networks_arr, networksString);
  }
}
```

bblanchon, 2022-07-31:
Hi @basementmedia2,

Please read [How to determine the capacity of the JsonDocument?](https://arduinojson.org/v6/how-to/determine-the-capacity-of-the-jsondocument/)

Best regards,
Benoit

basementmedia2, 2022-07-31:
Hi @bblanchon,
 
thank you for your answer.
I already decided for [Technique 3: set a memory budget]

but: What happens, if the set memory is not enough.
Let's say I chose the storage for the equivalent of about 20 networks and I'm in an environment where there are 23 networks. 

- Does ESP then crash because the memory overflows? Or
- Is the JSON Document then empty or invalid because it has no valid endpoint?
- Does the JSON document then contain 20 networks and the twenty-first is simply not stored.

Sorry for these beginner questions and thanks again for your patience ;-)

Best wishes
Daniel

bblanchon, 2022-08-02:
Hi Daniel,

When the [`JsonDocument`](https://arduinojson.org/v6/api/jsondocument/) overflows:

* the MCU does not crash,
* [`deserializeJson()`](https://arduinojson.org/v6/api/json/deserializejson/) returns [`DeserializationError::NoMemory`](https://arduinojson.org/v6/api/misc/deserializationerror/#deserializationerrornomemory),
* the [`JsonDocument`](https://arduinojson.org/v6/api/jsondocument/) is not flushed,
* [`JsonDocument::overflowed()`](https://arduinojson.org/v6/api/jsondocument/overflowed/) returns `true`.

Daniel, please help me improve the documentation by telling me where I should put this information so that users can find it easily. For example, where did you look before opening an issue?

Best regards,
Benoit

basementmedia2, 2022-08-02:
Hi,

i looked in your documentation (https://arduinojson.org/v6/how-to/determine-the-capacity-of-the-jsondocument/) 
But there i did not find information about what happanes, if JSON Document overflows.
Maybe an idea for placing this information would be below the section "What if I don’t know my input ahead of time?".
The section could be titled "What if the set capacity is not enough and JSON document overflows"?

From technic side it would be great if the deserializeJson() function would not return "NoMemory" but keep everything inside the document which "fits in" and keeps only the not fitting in data out -> don't know if this is possible and makes sense for everyone / every use case. Maybe this feature could be set to true.

I hope you understand what i mean. 
My english is not the best ;-)

Best wishes
Daniel

bblanchon, 2022-08-04:
Hi Daniel,

I added the section to the documentation.

I think the current behavior makes sense.
In your case, you just need to ignore the `NoMemory` error.

Best regards,
Benoit

closed 2022-08-22 by bblanchon