bblanchon/ArduinoJson #1783: Merge code maybe needs correction  : https://arduinojson.org/v6/how-to/merge-json-objects/

vortex314, 2022-07-30:
Code on the web page should maybe be adapted ? 
From : 
````
void merge(JsonVariant dst, JsonVariantConst src){
  if (src.is<JsonObject>()) {
    for (auto kvp : src.as<JsonObject>()){
      merge(dst.getOrAddMember(kvp.key()), kvp.value());
    }
  }
  else{
    dst.set(src);
  }
}
```
To : 
```
void merge(JsonVariant dst, JsonVariantConst src) {
  if (src.is<JsonObjectConst>()) {
    for (auto kvp : src.as<JsonObjectConst>()) {
      merge(dst.getOrAddMember(kvp.key()), kvp.value());
    }
  } else {
    dst.set(src);
  }
}
```
Otherwise the code didn't compile, thanks to Troubleshooter resolved : 
![image](https://user-images.githubusercontent.com/5508735/181917400-e375f0b8-ee18-420a-ae17-64a3475450bb.png)

vortex314, 2022-07-30:
When I do now a checkout 6.x , I got another message : 
```
/home/lieven/workspace/serial2redis/limero/linux/ConfigFile.cpp:83:21: error: ‘ArduinoJson::JsonVariant’ {aka ‘class ArduinoJson6194_71::VariantRef’} has no member named ‘getOrAddMember’
   83 |       deepMerge(dst.getOrAddMember(kvp.key()), kvp.value());
```

bblanchon, 2022-07-31:
Hi @vortex314,

Thank you very much for reporting this bug.
I just updated the page; please confirm that it works as expected.

FYI, ArduinoJson 6.20 removes the internal API: `getElement()`, `getMember()`, `getOrAddMember()`...
I should never have used this function here in the first place.
Sorry about that.

Best regards,
Benoit

vortex314, 2022-07-31:
Hi Benoit,
thanks for the swift reaction ! 
The merge function now works as expected.
No worries about the issue. 
I'm using your library for a lot of projects and use cases. In this case it was for merging configuration data. 
Thanks  a lot for all the continuous effort you keep putting in this software.
Lieven

closed 2022-07-31 by vortex314