bblanchon/ArduinoJson #1785: Much easier fix for min vs std::min isse

dberlin, 2022-08-03:
(It's not obvious where the website docs live in github, or i'd just submit a pull request)
The troubleshooter for the std::min vs arduino min macro (https://arduinojson.org/v6/troubleshooter/?utm_source=github&utm_medium=readme#compiletime/macro-min/success) says to disable std::string/std::stream support.

Just to flag there is actually a much easier fix that  does not require compiler flags and should work everywhere.

Before including ArduinoJson.h, do this:
```
#include <Arduino.h>
#undef min
```

min is guaranteed to be defined in C++ so we don't have to worry that we will remove the only definition (it's also supposed to be an inline function and not a macro, which is why it conflicts in the first place)

Including the header ourselves guarantees that all the macros are defined and that the header won't be reincluded by someone else.
`#undef min` then removes the conflicting macro definition.
This will work:
1. On all systems where someone uses ArduinoJSON
2. whether the min macro exists or not.
3. without removing support for std::*

Figured i would flag this.

bblanchon, 2022-08-04:
Hi @dberlin,

Thank you for this feedback.

The source of the website is, indeed, in a private repo.

Is it sufficient to `#undef min`? or do we also need to `#undef max`?

Best regards,
Benoit

dberlin, 2022-10-11:
At least as you have things now, you would not have to #undef max, because
ArduinoJson doesn't use max.
If that changed, yes, you would run smack into the same problem unless you
#undef max.

bblanchon, 2022-10-13:
ArduinoJson doesn't use `min` either, but the STL uses it.