bblanchon/ArduinoJson #1786: null returned when searching for boolean element that is set to false

Maurizio1, 2022-08-08:
Is there something I'm missing or is there a bug?

```
{
  "version" : "1.0.0.1",
  "boolean_value" : false,
  "another_boolean_value" : true,
  "bugtest" : "testbug"
}

JsonVariant result;
const char * key;

key = "boolean_value";
JsonVariant foundObject = result[key]; // this returns NULL
key = "another_boolean_value";
JsonVariant foundObject = result[key]; // this returns true
```

The error is systematic. I can successfully retrieve any type of element's value, but booleans when their value is 'false'. This happens in any JSON I tested.

P.S. If I use a JsonObject for result (JsonObject result;) the problem is still there.
 
Thank you in advance for any help.

bblanchon, 2022-08-09:
Hi @Maurizio1,

Here is a demo: https://wandbox.org/permlink/nLTnhqunyGmSJIes
As you can see, it correctly extracts `false` from the key `boolean_value`.

If you need more assistance, please provide an [MCVE](https://stackoverflow.com/help/minimal-reproducible-example) and include the [Troubleshooter](https://arduinojson.org/v6/troubleshooter/) report.

Best regards,
Benoit

Maurizio1, 2022-08-09:
> Hi @Maurizio1,
>
> Here is a demo: https://wandbox.org/permlink/nLTnhqunyGmSJIes As you can see, it correctly extracts `false` from the key `boolean_value`.
>
> If you need more assistance, please provide [...]

Hi Benoit,

thank you for the prompt answer and the demo.

As always, talking with someone opens the trap in which your mind has caught. :-)
Mine was that I did a test to verify if there was a search result or not and that test, now I see, obviously always failed for 'false' booleans.

**working** pseudo code:

```
  if (!foundObject.isNull()) {
      return foundObject;
  }
```

my **wrong** pseudo code was:

```
  if (foundObject) {
      return foundObject;
  }
```

which I took from the example:

**How to recursively search for a key?**

```
JsonVariant findNestedKey(JsonObject obj, const char* key) {
    JsonVariant foundObject = obj[key];
    if (foundObject)
        return foundObject;

    for (JsonPair pair : obj) {
        JsonVariant nestedObject = findNestedKey(pair.value(), key);
        if (nestedObject)
            return nestedObject;
    }

    return JsonVariant();
}
```

(I have emphasized it here as a suggestion for others)

Maybe you want to slightly change the example.

Thank you for your help and for this fantastic library!

Best regards,

Maurizio

bblanchon, 2022-08-10:
Hi Maurizio,

I updated the documentation as you suggested.
Thank you for sharing this.

Best regards,
Benoit

closed 2022-08-10 by bblanchon