bblanchon/ArduinoJson #1788: Setting volatile keyword in static json document

ramadhanafif, 2022-08-12:
**Description**
I am running a multi thread program on ESP32 using ESP freeRTOS. This program consists of two main thread, one of them collects data from sensor, and the other upload them to the internet. The variable that get passed around is a static json document. 

Since this is a multi-thread operation, I used volatile to avoid it getting removed by compiler optimization. However, I discover it wouldn't allow me to compile it with the following error:

error: passing 'volatile ArduinoJson6194_F1::StaticJsonDocument<512>' as 'this' argument discards qualifiers [-fpermissive] 

Currently, my workaround is by lowering compiler optimization level from -Os to -O2. Is there a better way to do this?

**Troubleshooter's report**
1. The issue happens at compile time
2. The error is not in the list

**Environment**
* Microcontroller: ESP32
* Core/Framework: ESP32 Arduino Core
* IDE: PlatformIO

**Reproduction code**
```c++
volatile StaticJsonDocument<512> doc;

```

bblanchon, 2022-08-13:
Hi @ramadhanafif,

`StaticJsonDocument` is not safe to use on concurrent threads, so I cannot mark its member as `volatile`

To work around this issue, you could push the sensor values to a queue and pull from the main thread.

Best regards,
Benoit

ramadhanafif, 2022-08-13:
Thanks @bblanchon for the answer, I figure this would be the case.

closed 2022-08-13 by ramadhanafif