bblanchon/ArduinoJson #1789: Send & receive Json on RPI-Arduino via Usb

Bingsitfx, 2022-08-20:
I want to do bidirectional serial communication between Raspi 3b and Arduino Mega, using usb. This is my situation, my arduino is connected to sensors & actuators, my arduino send json that contain "sensor data" to RPI, then RPI process and send json that contain "actuators data" to arduino.  

Arduino to RPI
StaticJsonDocument<256> doc1;
doc1["sensor1"] = 90
SerializeJson(doc1,Serial)

RPI to Arduino
StaticJsonDocument<256> doc2;
DeserializationError error = deserializeJson(doc2, Serial);
actuator1 = doc2["actuator1"]

Or can you give some example how to do that?

bblanchon, 2022-08-22:
Hi @Bingsitfx,

You should find some inspiration on this page: [How to do serial communication between two boards?](https://arduinojson.org/v6/how-to/do-serial-communication-between-two-boards/)

Best regards,
Benoit

Bingsitfx, 2022-08-22:
Thankyou

closed 2022-08-22 by Bingsitfx