bblanchon/ArduinoJson #1790: not working.

tbillion, 2022-08-22:
**Description**
/home/t/Arduino/libraries/ArduinoJson/src/ArduinoJson/Polyfills/pgmspace_generic.hpp: In instantiation of 'typename ArduinoJson6194_F1::enable_if<ArduinoJson6194_F1::is_pointer<T>::value, T>::type ArduinoJson6194_F1::pgm_read(const void*) [with T = const arduino::__FlashStringHelper*; typename ArduinoJson6194_F1::enable_if<ArduinoJson6194_F1::is_pointer<T>::value, T>::type = const arduino::__FlashStringHelper*]':
/home/t/Arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/DeserializationError.hpp:85:12:   required from here
/home/t/Arduino/libraries/RF24/RF24_config.h:206:38: error: 'const void*' is not a pointer-to-object type
  206 |             #define pgm_read_ptr(p) (*(p))
      |                                     ~^~~~~
exit status 1

**Troubleshooter's report**
1. The issue happens at compile time
2. The error is not in the list

**Environment**
* Microcontroller: pico
* Core/Framework: ear philhower most recent
* IDE: 1.8.16

**Reproduction code**
```c++
just adding the library throws the error , no actual code to perform actions has been added
```

**Remarks**
nope would just like it to work.

bblanchon, 2022-08-23:
Hi @tbillion,

> 'const void*' is not a pointer-to-object type

This error *is* in the list. Please look again.

Best regards,
Benoit

bblanchon, 2022-08-23:
Related to https://github.com/arduino/ArduinoCore-API/pull/118.
Duplicate of #1442, #1519, #1536, #1591, and #1676.

bblanchon, 2022-08-23:
In this case, it seems that the bug doesn't come from the core but from the RF24 library.
I'll open a PR.

bblanchon, 2022-09-03:
@2bndy5, here is a complete repro:

```c++
#include <RF24.h>
#include <ArduinoJson.h>

void setup() {}
void loop() {}
```

With RF24 v1.4.5, these four lines produce the following error on some Arduino cores:

| Core | Status |
|:---|:---|
| Adafruit SAMD | OK |
| Arduino AVR | OK |
| Arduino Mbed | Error |
| Arduino megaAVR | OK |
| Arduino SAMD | OK |
| ESP32 | OK |
| ESP8266 | OK |
| Raspberry Pi RP2040 | Error |
| STM32 | OK |
| STM32F1 | OK |
| STM32F4 | Error |
| Teensy 4 | OK |
| Teensy 3 | OK |
| Infineon XMC | Error |

2bndy5, 2022-09-03:
Thanks, that helps narrow down the cores, but the error in the OP doesn't seem to align with v1.4.5. I have a hunch at which line is the problem, but I just wanted to be sure.

tbillion, 2022-09-04:
1.4.3

2bndy5, 2022-09-05:
Avoid v1.4.4 because there is a known bug which was fixed in v1.4.5 (the latest). However, updating the lib won't likely fix this problem. I just wanted to know what version you were using so I could find the exact line that was causing the problem. Thank you for the response.

2bndy5, 2022-09-06:
I will let you know when the fix for this has been released (likely going to be v1.4.6) for the Arduino Library Manager... It should be soon, but I wanted to go through and fix the other `pgm_read_*` macros and add `printf_P` support to the arduino-pico core (+ ESP32/8266 cores).

closed 2022-09-08 by bblanchon

2bndy5, 2022-09-18:
I just released v1.4.6 which includes the fix for this. Please wait at least 24 hours for the release to get picked up by the Arduino Lib Manager, then update and enjoy...