bblanchon/ArduinoJson #1796: Remove elements from nested structure

3KUdelta, 2022-09-12:
Dear Benoit - many thanks for this great library. I am using ArduinoJson in many projects but now I really got stuck. I was searching all over but couldn't find any clue in the documents.
I am using V6 on an ESP32 in a WebRadio project.
The JsonDoc jStations is globally defined in setup():
```
DynamicJsonDocument jStations(2048);
.
.
int GetStationsFromSPIFFS(String path) {
  String Stations = ReadFromSPIFFS(path);
  DeserializationError err = deserializeJson(jStations, Stations);
  if (err) {
    Serial.print("DeserializeJson() download failed with code: ");
    Serial.println(err.f_str());
  }
  // Print the result for debugging
  // serializeJsonPretty(jStations, Serial);
  return jStations["station"].size();  // returns total amount of stations
}
```
jStations contains the following structure:
```
{
  "station": [
    {
      "stationname": "Swiss Pop",
      "url": "https://stream.srg-ssr.ch/m/rsp/mp3_128"
    },
    {
      "stationname": "Deluxe Easy",
      "url": "http://deluxe.hoerradar.de/deluxe-easy-mp3-hq"
    },
    {
      "stationname": "Absolut Chill",
      "url": "http://ais-sa5.cdnstream1.com/b05055_128mp3"
    },
    {
      "stationname": "Energy Hits",
      "url": "http://energyhits.ice.infomaniak.ch/energyhits-high.mp3"
    }
  ]
}
```
Now I wonder how I can delete element 3 (e.g. Absolut Chill and corresponding URL)? I tried many things including like this - but no success.
```
void delete_station(int station_no) {
  WebSerial.print("Station to be deleted: ");
  WebSerial.println(--station_no);            //array index starts with 0, list with 1

  // how to remove station_no?
  station_item["stationname"].remove("Absolut Chill");  // does not work

  String stations;     // check new structure
  serializeJson(jStations, stations);
  WebSerial.println(stations);
}
```
Appreciate your help. Marc

bblanchon, 2022-09-13:
Hi Marc,

Please see the examples in [JsonArray::remove()](https://arduinojson.org/v6/api/jsonarray/remove/#examples)

Best regards,
Benoit

3KUdelta, 2022-09-13:
Many thanks, Benoit. This made my day. 

"If you want to walk over water, you need to know where the stones are."

Marc

In case someone else searches for a similar case. Here is the working code.
```
void delete_station(int station_no) {
  WebSerial.print("Station to be deleted: ");
  WebSerial.println(--station_no);            //array index starts with 0, list with 1

  JsonArray delStation = jStations["station"];
  delStation.remove(station_no);

  String stations;
  serializeJson(jStations, stations);

  WriteToSPIFFS("/stations.json", stations);  // writing back to SPIFFS

  totalStations = GetStationsFromSPIFFS("/stations.json");  // reloading new structure
}
```

closed 2022-09-13 by 3KUdelta