bblanchon/ArduinoJson #1798: Issue between keyboard and chair: Unable to determine size of array

tijopr, 2022-09-13:
Hi thank you for your great work!

I am using ArduinoJson 6.19.4

I have a Json document produced from a call to openweather API. I am trying to determine the number of alerts in the document. I can determine if there are alerts in the Json by using containsKey("alerts"), but I am unable to determine how many alerts are in the Json as sometimes there is only 1 and other times there are 2. 

Code Snippet:
``
deserializeJson(doc, payload); 

  if (doc.containsKey("alerts")) {
    Serial.println("Got alerts");
  //Here I would like to determine how many alerts are present
  }
  else {
    Serial.println("No alerts");
  }
  Serial.println("# de keys in alerts:");`
``
I have tried to adapt an example from page 50 https://arduinojson.org/v6/doc/deserialization/ but it doesn't compile on my end.

``
  JsonArray miAlerta = doc.["alerts"]; 

  for (JsonObject value : miAlerta) {
    Serial.print(value["event"].as<char *>());
    Serial.print(value["description"].as<char*>());
}
I get this error: 
``
exit status 1
expected unqualified-id before '[' token
``

Here is the Json document

``
{"lat":18.4274,"lon":-67.1541,"timezone":"America/Puerto_Rico","timezone_offset":-14400,"current":{"dt":1662735043,"sunrise":1662718534,"sunset":1662763001,"temp":89.62,"feels_like":102.22,"pressure":1013,"humidity":69,"dew_point":78.08,"uvi":9.89,"clouds":43,"visibility":10000,"wind_speed":3.62,"wind_deg":55,"wind_gust":6.51,"weather":[{"id":802,"main":"Clouds","description":"scattered clouds","icon":"03d"}]},"alerts":[{"sender_name":"NWS San Juan (San Juan, Puerto Rico)","event":"Rip Current Statement","start":1662686040,"end":1662760800,"description":"...HIGH RIP CURRENT RISK REMAINS IN EFFECT THROUGH FRIDAY\nAFTERNOON...\n* WHAT...Dangerous rip currents expected due to breaking waves to\naround 6 feet.\n* WHERE...Beaches of northwestern Puerto Rico.\n* WHEN...Through Friday afternoon.\n* IMPACTS...Dangerous swimming conditions. Rip currents can\nsweep even the best swimmers away from shore into deeper\nwater.","tags":[]},{"sender_name":"NWS San Juan (San Juan, Puerto Rico)","event":"Heat Advisory","start":1662732000,"end":1662753600,"description":"...HEAT ADVISORY IN EFFECT FROM 10 AM THIS MORNING TO 4 PM AST\nTHIS AFTERNOON...\n* WHAT...Excessive heat with heat indices of 102-109 degrees\nexpected.\n* WHERE...Some lower elevations of northern-central, northwest and\nwest Puerto Rico.\n* WHEN...Between 10 AM and 4 PM AST this afternoon.\n* IMPACTS...High temperatures and humidity may cause heat\nillnesses to occur. Heat exhaustion likely with prolonged\nexposure. Heat stroke possible.","tags":["Extreme temperature value"]}]}
Pretty:
{
  "lat": 18.4274,
  "lon": -67.1541,
  "timezone": "America/Puerto_Rico",
  "timezone_offset": -14400,
  "current": {
    "dt": 1662735043,
    "sunrise": 1662718534,
    "sunset": 1662763001,
    "temp": 89.62,
    "feels_like": 102.22,
    "pressure": 1013,
    "humidity": 69,
    "dew_point": 78.08,
    "uvi": 9.89,
    "clouds": 43,
    "visibility": 10000,
    "wind_speed": 3.62,
    "wind_deg": 55,
    "wind_gust": 6.51,
    "weather": [
      {
        "id": 802,
        "main": "Clouds",
        "description": "scattered clouds",
        "icon": "03d"
      }
    ]
  },
  "alerts": [
    {
      "sender_name": "NWS San Juan (San Juan, Puerto Rico)",
      "event": "Rip Current Statement",
      "start": 1662686040,
      "end": 1662760800,
      "description": "...HIGH RIP CURRENT RISK REMAINS IN EFFECT THROUGH FRIDAY\nAFTERNOON...\n* WHAT...Dangerous rip currents expected due to breaking waves to\naround 6 feet.\n* WHERE...Beaches of northwestern Puerto Rico.\n* WHEN...Through Friday afternoon.\n* IMPACTS...Dangerous swimming conditions. Rip currents can\nsweep even the best swimmers away from shore into deeper\nwater.",
      "tags": []
    },
    {
      "sender_name": "NWS San Juan (San Juan, Puerto Rico)",
      "event": "Heat Advisory",
      "start": 1662732000,
      "end": 1662753600,
      "description": "...HEAT ADVISORY IN EFFECT FROM 10 AM THIS MORNING TO 4 PM AST\nTHIS AFTERNOON...\n* WHAT...Excessive heat with heat indices of 102-109 degrees\nexpected.\n* WHERE...Some lower elevations of northern-central, northwest and\nwest Puerto Rico.\n* WHEN...Between 10 AM and 4 PM AST this afternoon.\n* IMPACTS...High temperatures and humidity may cause heat\nillnesses to occur. Heat exhaustion likely with prolonged\nexposure. Heat stroke possible.",
      "tags": [
        "Extreme temperature value"
      ]
    }
  ]
}
``

bblanchon, 2022-09-14:
Hi @tijopr,

>  I am unable to determine how many alerts are in the Json

Like this:

```c++
size_t alertCount = doc["alerts"].size();
```

Best regards,
Benoit

tijopr, 2022-09-14:
Thanks for your prompt response.  It worked like a charm!