bblanchon/ArduinoJson #1799: ESP32S3 compile out of Arduino IDE and get some error

ajtwlsdlqp, 2022-09-22:
**Description**
Hi, 
I'm try to addapted "ArduinoJson-v6.19.4" to ESP32 S3 Board in not arduino IDE but eclipse IDE environment with esp-idf

almost success but i don't know why this error accrued
i think error accrued reason is that my eclipse IDE can not understand this type of code

include all arduino esp32 boards core lib and solve include path error
 - use ARDUINOJSON_ENABLE_ARDUINO_STRING 1
 - use ARDUINOJSON_ENABLE_ARDUINO_STREAM 1

and when i just try the build
Build complete (0 errors, 0 warnings): D:\SW_Work\build
Total time taken to build the project: 79,083 ms

no error, but still in error symbol in code "as<String>" or other "as<T>"  

if i just use default setup that both of enable define as "0",
the other error also accrued and build can not finish.

please help me understand this situation
best regards,

**Troubleshooter's report**
1. The issue happens at compile time
2. The error is not in the list

**Environment**
* Microcontroller: ESP32S3
* Core/Framework: -
* IDE: Eclipse IDE for C/C++ Developers [2022-06 (4.24.0)]

**Reproduction code**
```c++
try to run 'https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot" examples
and get red underline (error symbol) in this code at ".as<String>"
 

bool UniversalTelegramBot::getMe() {
  String response = sendGetToTelegram(BOT_CMD("getMe")); // receive reply from telegram.org
  DynamicJsonDocument doc(maxMessageLength);
  DeserializationError error = deserializeJson(doc, ZERO_COPY(response));
  closeClient();

  if (!error) {
    if (doc.containsKey("result")) {
      name = doc["result"]["first_name"]._as<String>();_
      userName = doc["result"]["username"]._as<String>();_
      return true;
    }
  }

  return false;
}

```

bblanchon, 2022-09-23:
Hi @ajtwlsdlqp,

What's the error message?

Best regards,
Benoit

ajtwlsdlqp, 2022-09-23:
Hello, @bblanchon ,
Thanks for help.

i can check error message

when i use 
CODE : name = doc["result"]["first_name"].as<WString>();
ERROR MESSAGE : Method 'as' could not be resolved

i think error when accessing an object by index as a two-dimensional array.

Best regards,

bblanchon, 2022-09-23:
But it works when you compile the code, right?

Do you have the same error when you do `doc["result"]["first_name"].as<const char*>();`?

Is this the only error on the page?

ajtwlsdlqp, 2022-09-23:
But it works when you compile the code, right? -> yes build is work.
doc["result"]["first_name"].as<const char*>();     -> also make same error

Is this the only error on the page?
-> 
actually  `DeserializationError error = deserializeJson(doc, ZERO_COPY(response));`
this code also make error but i just block this function and build 
make my code -> error check -> allways none;

thanks for interest

bblanchon, 2022-09-23:
This really sounds like a problem with IDE itself, so there is probably nothing we can do.
I'll try to find some time to install Eclipse and see for myself.

ajtwlsdlqp, 2022-09-26:
@bblanchon 
Thanks for your interest in the question.
Hope you have a nice day.

closed 2022-09-26 by ajtwlsdlqp

bblanchon, 2022-09-26:
Hi @ajtwlsdlqp,

I manage to reproduce the issue, but as far as I can tell, it only occurs when the type passed to `as<T>()` is unresolved.

See this screen capture for example:

[image]

As you can see, there is an error when using the unresolved type `foobar`, but no error with `std::string`.

Best regards,
Benoit

ajtwlsdlqp, 2022-09-26:
@bblanchon

oh, i see. 
i was found that issue, during convert arduino library [Univertial Telegram Bot] to esp32 project 

could you see my screenshot also?
[image]

when i use as<T> in one dimension JsonDocument no error occurred
but when use as<T> function in two or more dimension IDE return error

tnanks for support 
Best regards,