bblanchon/ArduinoJson #1801: Question About Filtering

unmanagedtn, 2022-09-28:
Below I am trying to filter out the "flowSegmentData" to get the "currentSpeed" but the data returned for "currentSpeed" is null

What could I be doing wrong... I am not that great of a programmer so  I know I must be butchering it up some place...

`StaticJsonDocument<24576> filter;
 filter["flowSegmentData"][0]["currentSpeed"] = true;
 

  // Deserialize the document
 StaticJsonDocument<300> doc;
 //deserializeJson(doc, client, DeserializationOption::Filter(filter));
//DynamicJsonDocument doc(24576);

DeserializationError error = deserializeJson(doc, client, DeserializationOption::Filter(filter));
serializeJsonPretty(doc, Serial);
//DeserializationError error = deserializeJson(doc, client);

if (error) {
  Serial.print(F("deserializeJson() failed: "));
  Serial.println(error.f_str());
  return;
}

JsonObject flowSegmentData = doc["flowSegmentData"];
const char* flowSegmentData_frc = flowSegmentData["frc"]; // "FRC2"
int flowSegmentData_currentSpeed = flowSegmentData["currentSpeed"]; // 51
int flowSegmentData_freeFlowSpeed = flowSegmentData["freeFlowSpeed"]; // 51
int flowSegmentData_currentTravelTime = flowSegmentData["currentTravelTime"]; // 956
int flowSegmentData_freeFlowTravelTime = flowSegmentData["freeFlowTravelTime"]; // 956
int flowSegmentData_confidence = flowSegmentData["confidence"]; // 1
bool flowSegmentData_roadClosure = flowSegmentData["roadClosure"]; // false

for (JsonObject flowSegmentData_coordinates_coordinate_item : flowSegmentData["coordinates"]["coordinate"].as<JsonArray>()) {

  double flowSegmentData_coordinates_coordinate_item_latitude = flowSegmentData_coordinates_coordinate_item["latitude"];
  double flowSegmentData_coordinates_coordinate_item_longitude = flowSegmentData_coordinates_coordinate_item["longitude"];

}

const char* flowSegmentData_version = flowSegmentData["@version"]; // "traffic-service-flow 1.0.060"`

unmanagedtn, 2022-09-28:
`{"flowSegmentData":{"frc":"FRC2","currentSpeed":51,"freeFlowSpeed":51,"currentTravelTime":956,"freeFlowTravelTime":956,"confidence":1,"roadClosure":false,"coordinates":{"coordinate":[{"latitude":0,"longitude":-0},
` the data is repeating so I removed it also has gps data in it and had to remove the remainder
?
and from what I am seeing is that I cant filter before the first [ is that correct?

I need to filter as I am running out of memory if I do not ...

bblanchon, 2022-09-29:
Hi @unmanagedtn,

If your input starts with: 

```json
{"flowSegmentData":{"frc":"FRC2","currentSpeed":51,
```

Then the filter must be something like:

```json
{"flowSegmentData":{"currentSpeed":true}}
```

not 

```json
{"flowSegmentData":[{"currentSpeed":true}]}
```

Also the size of your `JsonDocument`s look wrong, please use the code generated by the [Assistant](https://arduinojson.org/v6/assistant/).

Best regards,
Benoit

unmanagedtn, 2022-09-29:
> Hi @unmanagedtn <https://github.com/unmanagedtn>,
>
> If your input starts with:
>
> {"flowSegmentData":{"frc":"FRC2","currentSpeed":51,
>
> Then the filter must be something like:
>
> {"flowSegmentData":{"currentSpeed":true}}
>
> not
>
> {"flowSegmentData":[{"currentSpeed":true}]}
>
> Also the size of your JsonDocuments look wrong, please use the code
> generated by the Assistant <https://arduinojson.org/v6/assistant/>.
>
> Best regards,
> Benoit

I was under the impression that you have to have two sizes one for the
filter and one for the parse of the data in the doc is that correct?

unmanagedtn, 2022-09-29:
I guess i dont understand the example is listed as 

` StaticJsonDocument<200> filter;
  filter["list"][0]["dt"] = true;
  filter["list"][0]["main"]["temp"] = true;`

unmanagedtn, 2022-09-30:
**THANK YOU!** 

`StaticJsonDocument<64> filter;
filter["flowSegmentData"]["currentSpeed"] = true;
 

StaticJsonDocument<128> doc;

DeserializationError error = deserializeJson(doc, client, DeserializationOption::Filter(filter));
serializeJsonPretty(doc, Serial);
if (error) {
  Serial.print(F("deserializeJson() failed: "));
  Serial.println(error.f_str());
  return;
}

int flowSegmentData_currentSpeed = doc["flowSegmentData"]["currentSpeed"]; // 51

Serial.print ("Speed: ");
Serial.println(flowSegmentData_currentSpeed);
}
`

Gets this output ...

`WiFi connected
IP address: 
192.168.88.63
{
  "flowSegmentData": {
    "currentSpeed": 51
  }
}Speed: 51
`

So the project is to take a fake traffic light and show the traffic speed on the highway so I can tell early if traffic is bad before I am late to work .... :) Do you have a way to tip ?

closed 2022-09-30 by unmanagedtn

bblanchon, 2022-09-30:
Hi @unmanagedtn,

I'm glad you solved your issue!
If you want to send me a tip, you can either [purchase my book](https://arduinojson.org/book/), [make a donation on PayPal](https://donate.benoitblanchon.fr/), or [become a sponsor](https://github.com/sponsors/bblanchon).

Best regards,
Benoit