bblanchon/ArduinoJson #1803: Corrupted values

michaloles, 2022-09-28:
**Describe the bug**  
I am deserializing two json config files. For first one everything works perfectly. However when I deserialize second one it sometimes overwrite config object readed from first file. However when I removed two rows from second file I didn't saw this issue. I check for sollutions in ArduinoJson documentation, tried some things mentioned there like clear or garbageCollect json object, using separate JsonDocuments, passing them as argument so they live long enought etc but it still not work correctly and I am out of idea why.

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  
1. The issue happens at run time
2. The issue concerns deserialization
3. <code>deserializeJson()</code> returns <code>InvalidInput</code>
4. Input comes from a file
5. <code>jsonlint</code> says the document is valid
6. Input's first byte doesn't suggest a BOM

**Environment**  
Here is the environment that I used:
* Microcontroller: ESP32
* Core/runtime: espressif32@5.0.0
* IDE: Platform IO

**Reproduction**  
Here is a small snippet that reproduces the issue.

```c++
void setup(){
  readSystemConfiguration(); //Read first file
  readDeviceConfiguration(); //Read second file
}

void readSystemConfiguration(){
  Log.verboseln(F("readSystemConfiguration call"));

  DynamicJsonDocument doc(5000);
  fileSystem.loadConfiguration("/systemConfig.json", doc);
  systemConfig.fromJson(doc.as<JsonObject>());
  doc.clear();
}

void readDeviceConfiguration(){
  Log.verboseln(F("readDeviceConfiguration call"));

  DynamicJsonDocument doc(8000); 
  fileSystem.loadConfiguration("/deviceConfig.json", doc);

  Log.verboseln(F("SystemConfig is correct here: %s"), systemConfig.wifiName); //Until here everything works 
  deviceConfig.fromJson(doc.as<JsonVariant>());
  Log.verboseln(F("At this point system config is incorrect: %s"), systemConfig.wifiName); //Here content is already broken 
  doc.clear();
}

//fileSystem.loadConfiguration
void loadConfiguration(const char *filename, DynamicJsonDocument &doc) {
  File file = SPIFFS.open(filename, "r+");

  // Deserialize the JSON document
  DeserializationError error = deserializeJson(doc, file);
  if (error){
    Log.errorln(F("Failed to read file, using default configuration %s"), error.c_str());
  }
  file.close();
}

//deviceConfig.fromJson
void fromJson(const JsonVariant& json){
        JsonArray array = json["devices"].as<JsonArray>();
        mpptType = json["mpptType"];
        int i=0;
        for(JsonObject item: array){
            device[i].fromJson(item); //This call fromJson from SingleDeviceConfig
            i++;
        }
    }

class SingleDeviceConfig {

   public:
  
    char name[32];
    char mapping[20];
    int type2;
    int default2;
    bool connected;

    void fromJson(const JsonObject& json) {
        strlcpy(name, json["name"], sizeof(name));
        strlcpy(mapping, json["mapping"], sizeof(mapping));
        type2 = json["type"];
        default2 = json["default"];
        connected = json["connected"];
    }
};
```

systemConfig.json file:
```
{
    "wifiName": "MyWifiName",
    "wifiPassword": "wifiPassword",
    "connectToExistingWifi": false,
    "dhcp": true,
    "ip": "192.168.4.1",
    "gateway": "192.168.4.1",
    "subnet": "255.255.255.0",
    "user": "",
    "password": "",
    "phone": "",
    "pin": ""
}
```

deviceConfig.json file:
```
{"devices": [
    {"name": "Hotel battery", "mapping": "mainBattery", "type": 1, "connected": 1},
    {"name": "Running battery", "mapping": "additionalBattery",  "type": 2, "connected": 1},
    {"name": "Fresh water", "mapping": "waterSensor1", "type": 1, "connected": 1},
    {"name": "Grey water", "mapping": "waterSensor2", "type": 1, "connected": 1},
    {"name": "Light", "mapping": "device1", "type": 2, "connected": 1},
    {"name": "Fridge", "mapping": "device2",  "type": 8, "connected": 1},
    {"name": "Water pump", "mapping": "device3",  "type": 1, "connected": 1},
    {"name": "Water valve", "mapping": "device4",  "type": 7, "connected": 1},
    {"name": "Fan", "mapping": "device5",  "type": 4, "connected": 1},
    {"name": "Heating", "mapping": "device6", "type": 3, "connected": 1},
    {"name": "Power Inverter", "mapping": "device7",  "type": 9, "connected": 1},
    {"name": "Move sensor", "mapping": "sensor1",  "type": 4, "connected": 1},
    {"name": "Gas sensor", "mapping": "sensor2",  "type": 5, "connected": 1},
    {"name": "Flood sensor", "mapping": "sensor3",  "type": 6, "connected": 1},
    {"name": "Door sensor", "mapping": "sensor4",  "type": 7, "connected": 1},
    {"name": "Outside temp.", "mapping": "extTemp1",  "type": 1, "connected": 1},
    {"name": "Grey water temp.", "mapping": "extTemp2",  "type": 1, "connected": 1},
    {"name": "Custom temp. 1", "mapping": "extTemp3",  "type": 1, "connected": 0},
    {"name": "Custom temp. 2", "mapping": "extTemp4",  "type": 1, "connected": 0}
], 
    "mpptType": 1
}
```

**Program output**  

Expected output:
```
V: SystemConfig is correct here: MyWifiName
V: At this point system config is incorrec: MyWifiName
```
When I remove some rows from deviceConfig.json I can get this output. Of course I tried to increase json document size but it didn't help.

Actual output:

```
V: SystemConfig is correct here: MyWifiName
V: At this point system config is incorrec: MyWifiNaextTemp2
```
Looks like somehow it joined wifiName (without two last chars) from first file with mapping value from second file.

bblanchon, 2022-09-29:
Hi @michaloles,

The code seems fine at first glance.
The only thing I can think of is an overflow or an incorrect allocation of the `device` array.

Please provide an [MCVE](https://stackoverflow.com/help/mcve) (max 50 lines), so I can try too.

Best regards,
Benoit

michaloles, 2022-09-30:
Hi @bblanchon,
Thank you for help. Here is MCVE: [https://github.com/michaloles/arduino_json_issue/tree/main/src](https://github.com/michaloles/arduino_json_issue/tree/main/src)

Best regards,
Michal Oles

bblanchon, 2022-10-05:
Hi Michal,

I looked at the code you posted, and as I suspected, the problem seems to be caused by an overflow of the `device` array (19 elements for a capacity of 15).

Best regards,
Benoit

michaloles, 2022-10-07:
Hi @bblanchon 
You are perfectly right. That seems to fix the issue. I have no idea how I could miss that bug. Thank you for help and sorry for inconvinience.

Best regards,
Michal Oles
Michal Oles

closed 2022-10-07 by michaloles

bblanchon, 2022-10-08:
You're welcome, Michal!
Thank you for using ArduinoJson.
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink: