bblanchon/ArduinoJson #1808: YAML to JsonObject

tobozo, 2022-10-04:
hey @bblanchon and thanks for your awesome library !

[Ref.](https://twitter.com/BenoitBlanchon/status/1576609444086968320)
> given the differences/similarities between JSON and YAML, how far is serializeJsonPretty from producing yml output, and how hard would it be to produce an ArduinoYAML library based on your awesome work ?

your answer:
> serializeYaml() would be easy, but deserializeYaml() would require a lot of work!

`deserializeYaml()` is now implemented: https://github.com/tobozo/esp32-yaml

I wish I could produce the `serializeYaml()` and `serializeYmlPretty()` for ArduinoJson too but the level of C++ in this library is way beyond my skills.

tobozo, 2022-10-07:
thanks for your helpful feedback and marvelous documentation :+1: 

I came up with this recursive function and got a satisfying result, did I miss anything?

```cpp
enum JNestingType_t { NONE, SEQ_KEY, MAP_KEY };

#define indent(x) std::string(x*2, ' ').c_str()

void serializeYml( JsonVariant root, Stream &out, int depth_level=0, JNestingType_t nt=NONE )
{
  int parent_level = depth_level>0?depth_level-1:0;

  if (root.is<JsonArray>()) {
    JsonArray array = root;
    for( int i=0; i<array.size(); i++ ) {
      size_t child_depth_level = array[i].is<JsonObject>() ? depth_level+1 : depth_level-1;
      serializeYml(array[i], out, child_depth_level, SEQ_KEY);
    }
  } else if (root.is<JsonObject>()) {
    JsonObject object = root;
    int i = 0;
    for (JsonPair pair : object) {
      const char* _indent = (i==0 && nt==SEQ_KEY) ? indent(parent_level) : indent(depth_level);
      const char* _prefix = (i==0 && nt==SEQ_KEY) ? "- " : "";
      out.printf("\n%s%s%s: ", _indent, _prefix, pair.key().c_str() );
      serializeYml( pair.value(), out, depth_level+1, MAP_KEY );
      i++;
    }
  } else if( !root.isNull() ) {
    switch(nt) {
      case SEQ_KEY: out.printf("\n%s%s%s",  indent(depth_level), "- ", root.as<String>().c_str() ); break;
      default:  out.printf("%s",  root.as<String>().c_str() ); break;
    }
  } else {
    out.println("Error, unhandled type");
  }
}
```

bblanchon, 2022-10-07:
Yep, that's the idea.

tobozo, 2022-10-07:
Since most data processing softwares bundle their own codecs when exporting to other formats I mostly wrote this function for ArduinoJson (e.g. as an extra example), but just noticed the label change so I guess it's too much out of scope?

Please let me know if you want a PR for this, I'll add the function to my esp32-yaml library anyway.

thanks again for your support!

closed 2022-10-07 by tobozo