bblanchon/ArduinoJson #1809: Filter size

enjoyneering, 2022-10-07:
**Describe the issue**  
Hi bblanchon,

I have a question about the filter size. In the example below, do all 7 filters use one block of 256 bytes or does each filter use 256 bytes in memory (256 x 7 = 1792 bytes in RAM)? Is there any correlation between "filter" size and "doc" size? Any help would be appreciated.

**Environment**  
Here is the environment that I'm using':
* Microconroller: [e.g. ESP8266]
* Core/runtime: [e.g. ESP8266 core for Arduino v3.0.2]
* IDE: [e.g. Arduino IDE 1.8.16]

```c++
DynamicJsonDocument filter(256);

filter["weather"][0]["id"]   = true;
filter["weather"][0]["icon"] = true;
filter["main"]["temp"]       = true;
filter["main"]["pressure"]   = true;
filter["main"]["humidity"]   = true;
filter["sys"]["sunrise"]     = true;
filter["sys"]["sunset"]      = true;

DynamicJsonDocument doc(512);

deserializeJson(doc, client DeserializationOption::Filter(filter));
```

bblanchon, 2022-10-08:
Hi @enjoyneering,

The "filter" document is a regular `JsonDocument` and obeys the same sizing rules.
I don't understand why you would think that the size is multiplied by 7, but it's not.

As usual, use the [ArduinoJson Assistant](https://arduinojson.org/v6/assistant/) to compute the right capacity.
You can either use the "Serialize" mode and enter the filter in the "Output" field; in that case, you'll see the required capacity in Step 3, as with any `JsonDocument`
Or you can use the "Deserialize and filter" mode and enter the filter in the "Filter" field; in that case, you'll see the required capacity in the sample program.
In both cases, you should get a size of 176 bytes for this example.

Best regards,
Benoit

enjoyneering, 2022-10-08:
got it. thank you for the answer.

closed 2022-10-08 by enjoyneering