bblanchon/ArduinoJson #1810: No such file or directory

KheireddineAzzez, 2022-10-07:
**Describe the issue**  
Actually I have used ArduinoJson but there was a problem has been accrued. 

**Troubleshooter report**  
In file included from c:\users\kheir\documents\arduino\libraries\arduino-ddp\libs\arduinojson\src/ArduinoJson.hpp:24:0,
                 from c:\users\kheir\documents\arduino\libraries\arduino-ddp\libs\arduinojson\src/ArduinoJson.h:9,
                 from c:\users\kheir\documents\arduino\libraries\arduino-ddp\libs\arduinojson\arduinojson.h:5,
                 from c:\Users\kheir\Documents\Arduino\libraries\Arduino-DDP/DDP.h:14,
                 from C:\Users\kheir\Downloads\ss\thesis-sensor-master\pwm\pwm.ino:1:
c:\users\kheir\documents\arduino\libraries\arduino-ddp\libs\arduinojson\src/ArduinoJson/Array/ArrayRef.hpp:7:10: fatal error: ArduinoJson/Array/ArrayFunctions.hpp: No such file or directory
 #include <ArduinoJson/Array/ArrayFunctions.hpp>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1

**Environment**  
Here is the environment that I'm using':
* Core/runtime: Arduino Mega or Mega 2560
* IDE:  2.0.0

**Reproduction**  
[image]
```c++
DynamicJsonDocument doc(1024);

DeserializationError error = deserializeJson(doc, "{\"hello\":\"world\"}");

// insert code here
```

**Program output**  
If relevant, include the program output.

Expected output:

```
[insert expected output here]
```

Actual output:

```
[insert actual output here]
```

bblanchon, 2022-10-08:
Hi @KheireddineAzzez,

I cannot reproduce the problem.
Can you try to reinstall the library?
Can you try to move `#include <ArduinoJson.h>` at the top?

Best regards,
Benoit

KheireddineAzzez, 2022-10-08:
Actually, I am using a library  called [DDP-arduino](Arduino-DDP)  and I want to know what is the actual  problem. Because   there is message appeared, that told me I had to upgrade my JsonBuffer  from 5 to 6. Would you do a favor  for me .
Can you check please  the problem, because I need this for my thesis.

Thanks.

HamzaHajeir, 2022-10-30:
Hi there,
I have the exact issue but using CMake.

```
In file included from F:/Projects/SmartHome/Smart_Solutions/cppTest/cppTest2/libs/ArduinoJson/src/ArduinoJson.hpp:24:0,
                 from F:/Projects/SmartHome/Smart_Solutions/cppTest/cppTest2/libs/ArduinoJson/src/ArduinoJson.h:9,
                 from F:/Projects/SmartHome/Smart_Solutions/cppTest/cppTest2/libs/ArduinoJson/ArduinoJson.h:5,
                 from F:\Projects\SmartHome\Smart_Solutions\cppTest\cppTest2\libs\Framework\Formatter.cpp:5:
F:/Projects/SmartHome/Smart_Solutions/cppTest/cppTest2/libs/ArduinoJson/src/ArduinoJson/Array/ArrayRef.hpp:7:48: fatal error: ArduinoJson/Array/ArrayFunctions.hpp: No such file or directory
 #include <ArduinoJson/Array/ArrayFunctions.hpp>
                                                ^
```

The Caller `CMakeLists.txt` file includes the library by:
```
target_include_directories(framework PUBLIC ${ARDUINOJSON_DIR}) 
target_link_libraries(framework PUBLIC ArduinoJson)
```

bblanchon, 2022-10-31:
Hi @HamzaHajeir,

You should not get this error if you follow [the recommended CMake integration](https://arduinojson.org/v6/how-to/use-arduinojson-with-cmake/).
Alternatively, you can download the [single header version](https://github.com/bblanchon/ArduinoJson/releases/download/v6.19.4/ArduinoJson-v6.19.4.h), which eliminates the problem entirely.

Best regards,
Benoit

HamzaHajeir, 2022-10-31:
Hi Benoit

I'm very new to CMake (Started learning it two days ago), so the
recommendations you've provided would help much.

Thank you again,
Hamzah

bblanchon, 2022-11-03:
@KheireddineAzzez; did you solve your problem? Can I close this issue?

HamzaHajeir, 2022-11-03:
@bblanchon FWIW, I've used the first method and it works perfectly fine!

closed 2022-11-26 by bblanchon