bblanchon/ArduinoJson #1811: reclaiming memory by calling destructor

mars000, 2022-10-09:
**Describe the issue**  
I am parsing a faily lengthly json string. With a function I have the following code snippet:
      ```
         
                StaticJsonDocument<maxLengthBusinessRuleJsonMessage> doc;  // maxLengthBusinessRuleJsonMessage = 2900
		
		DeserializationError error = deserializeJson(doc, payload, length);

		if (error)
		{
			Serial.print(F("deserializeJson() failed: "));
			Serial.println(error.f_str());
			return;
		}
		else
		{
		  // all is fine...
		}
    	        serializeJsonPretty(doc, Serial);
		JsonArray docArrayReference = doc.as<JsonArray>();

```
Further on within the same function I am calling the following in the hope it will call the destructor and return all stack memory back.
`doc.~StaticJsonDocument();`

Is this correct method to destroy doc and reclaim all the stack memory allocated to it ?
I'm not sure its working....can I somehow validate it ?
Is the stack memory returned immediately ?

**Troubleshooter report**  
2022-10-09 20:14:51 --> Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
2022-10-09 20:14:51 --> Debug exception reason: Stack canary watchpoint triggered (loopTask) 
2022-10-09 20:14:51 --> Core  1 register dump:
2022-10-09 20:14:51 --> PC      : 0x4008e47a  PS      : 0x00060736  A0      : 0x40084e7e  A1      : 0x3ffb0880  
2022-10-09 20:14:51 --> A2      : 0x00000000  A3      : 0x40082cc4  A4    
2022-10-09 20:14:51 -->   : 0x00000000  A5      : 0x3ffbf148  
2022-10-09 20:14:51 --> A6      : 0x3ffc6c5c  A7      : 0x00063000  A8      : 0x00000102  A9      : 0x3ffb0920  
2022-10-09 20:14:51 --> A10     : 0x3ffc6b70  A11     : 0x00000001  A12     : 0x3ffc6b74  A13     : 0x00060723  
2022-10-09 20:14:51 --> A14     : 0x00060720  A15     : 0x00000001  SAR     : 0x00000017  EXCCAUSE: 0x00000001  
2022-10-09 20:14:51 --> EXCVADDR: 0x00000000  LBEG    : 0x4008a759  LEND    : 0x4008a769  LCOUNT  : 0xfffffffe  

**Environment**  
ESP32 LOLIN D32
VSC IDE
ArduinioJson Library version: 6.19.4
**Reproduction**  
Here is a small snippet that demonstrate the problem.

```c++
DynamicJsonDocument doc(1024);

DeserializationError error = deserializeJson(doc, "{\"hello\":\"world\"}");

// insert code here
```

**Program output**  
If relevant, include the program output.

Expected output:

```
[insert expected output here]
```

Actual output:

```
[insert actual output here]
```

bblanchon, 2022-10-11:
Hi @mars000,

This `StaticJsonDocument` is an [automatic variable](https://en.wikipedia.org/wiki/Automatic_variable), so the memory is reclaimed as soon as the execution pointer leaves `doc`'s scope.
For example:

```c++
{
  StaticJsonDocument<128> doc;
  // some code that uses doc
}
// memory is now reclaimed!
```

Best regards,
Benoit

mars000, 2022-10-12:
Thank you Benoit.

bblanchon, 2022-10-12:
You're welcome, @mars000.
Thank you for using ArduinoJson.
Don't forget to [cast a star](https://github.com/bblanchon/ArduinoJson/stargazers) to support the project :wink:

closed 2022-10-12 by bblanchon