bblanchon/ArduinoJson #1812: ArduinoJson.h error: if (_destination->concat(_buffer))

gck303, 2022-10-11:
**Description**
I would expect the code to complile!

**Troubleshooter's report**
1. The issue happens at compile time
2. The error is not in the list

**Environment**
* Microcontroller: Teensy 4.1
* Core/Framework: Teensyduino 1.57
* IDE: 1.8.19

**Reproduction code**
```c++
#include <ArduinoJson.h>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

**Remarks**
I cannot get the ArduinoJson.h include to compile correctly. I get these errors:

In file included from C:\Users\gck.CAMLEY\Documents\libraries\ArduinoJson\src/ArduinoJson/Serialization/Writer.hpp:38:0,
                 from C:\Users\gck.CAMLEY\Documents\libraries\ArduinoJson\src/ArduinoJson/Serialization/serialize.hpp:7,
                 from C:\Users\gck.CAMLEY\Documents\libraries\ArduinoJson\src/ArduinoJson/Json/JsonSerializer.hpp:10,
                 from C:\Users\gck.CAMLEY\Documents\libraries\ArduinoJson\src/ArduinoJson/Variant/ConverterImpl.hpp:7,
                 from C:\Users\gck.CAMLEY\Documents\libraries\ArduinoJson\src/ArduinoJson.hpp:37,
                 from C:\Users\gck.CAMLEY\Documents\libraries\ArduinoJson\src/ArduinoJson.h:9,
                 from C:\Users\gck.CAMLEY\Documents\Arduino\sketch_oct11b\sketch_oct11b.ino:1:
C:\Users\gck.CAMLEY\Documents\libraries\ArduinoJson\src/ArduinoJson/Serialization/Writers/ArduinoStringWriter.hpp: In member function 'size_t ArduinoJson6194_1::Writer<String, void>::flush()':
C:\Users\gck.CAMLEY\Documents\libraries\ArduinoJson\src/ArduinoJson/Serialization/Writers/ArduinoStringWriter.hpp:42:29: error: could not convert '((ArduinoJson6194_1::Writer<String, void>*)this)->ArduinoJson6194_1::Writer<String, void>::_destination->String::concat(((const char*)((char*)(&((ArduinoJson6194_1::Writer<String, void>*)this)->ArduinoJson6194_1::Writer<String, void>::_buffer))))' from 'String' to 'bool'
     if (_destination->concat(_buffer))
         ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~
Error compiling for board Teensy++ 2.0.

bblanchon, 2022-10-11:
Hi @gck303,

Unfortunately, Teensy's core doesn't implement Arduino's `String` class correctly.
See https://github.com/PaulStoffregen/cores/issues/561

It works on recent versions of Teensy because they [added a conversion from `String` to `bool`](https://github.com/PaulStoffregen/cores/commit/253da1599de5a74be1b219a5a34d23366ca0c2c0), but this change was never backported to Teensy 2.
Given the silent reaction I received, I didn't try to open a PR.

As a workaround, you can disable the support for Arduino's `String` class by defining [ARDUINOJSON_ENABLE_ARDUINO_STRING](https://arduinojson.org/v6/api/config/enable_arduino_string/) to 0.

Best regards,
Benoit

gck303, 2022-10-11:
Wow. Thank you. 

I cannot image how long I would have taken to find this. 

I have added the line "#define ARDUINOJSON_ENABLE_ARDUINO_STRING 0" and it compiles now.

closed 2022-10-11 by gck303