bblanchon/ArduinoJson #1814: MsgPack EEPROM Stream Serialization Issues

jrhodnik, 2022-10-14:
**Describe the issue**  
When serializing MsgPack to an EEPROM stream, it doesn't seem to deserialize properly. However, with the same code, using JSON serialization things work fine.

**Troubleshooter report**  
Here is the report generated by the [ArduinoJson Troubleshooter](https://arduinojson.org/v6/troubleshooter/):  
1. The issue happens at run time
2. The issue concerns deserialization
3. <code>deserializeJson()</code> returns <code>Ok</code>
4. Program fails to extract values from the <code>JsonDocument</code>
5. Program doesn't use <code>DeserializationOption::Filter</code>
6. Input doesn't start with quote
7. Issue is not a confusion between array and object

**Environment**  
Here is the environment that I'm using':
* Microconroller: ESP32-WROOM-32
* Core/runtime: Arduino
* IDE: Arduino IDE 1.8.19

**Reproduction**  
See the code below.

```c++

#include <EEPROM.h>
#include <ArduinoJson.h>
#include <StreamUtils.h>

StaticJsonDocument<256> config;

void ReadConfig() {
  EepromStream eepromStream(0, 255);
  auto err = deserializeMsgPack(config, eepromStream);
  
  if (err) {
      Serial.print(F("deserializeMsgPack() failed: "));
      Serial.println(err.c_str());
  }
}

void WriteConfig() {
  EepromStream eepromStream(0, 255);
  serializeMsgPack(config, eepromStream);
  
  eepromStream.flush();
}

void setup() {
  Serial.begin(115200);
  EEPROM.begin(255);

  ReadConfig();

  if (config["bootCount"].isNull())
    Serial.println("No boot counts!");

  uint32_t bootCount = config["bootCount"].as<uint32_t>();
  printf("Boot count: %d\n", bootCount);

  config["bootCount"] = bootCount + 1;

  WriteConfig();
}

void loop() {
}
```

**Program output**  
Always returns: 
```
No boot counts!
Boot count: 0
```

Expected output: Incrementing boot counts.

bblanchon, 2022-10-15:
Hi @jrhodnik,

It took me quite some to find the problem because the bug is so subtle.

1. You first ran this program with `serializeJson()`, so the EEPROM contains `{"bootCount":1}`
2. Then, you changed the code to use `deserializeMsgPack()`
3. `deserializeMsgPack()` reads the first character from the EEPROM: `{`
4. `{` has the hexadecimal value `7B`, which is the MsgPack encoding for the integer `72`
5. `deserializeMsgPack()` happily saves `72` in the document root and stops reading
6. No error is returned because this was a valid MsgPack document
7. Your program detects that the document doesn't contain the `bootCount` key.
  Indeed, the document is not an object but an integer with the value `72`
8. Your program tries to set the `bootCount` member to `1`, but it fails.
  Indeed, integers don't have members; only objects do.
9. Finally `serializeMsgPack()` writes `7B` back into the EEPROM.
10. Reboot and go to 1.

To fix this error, you must reset the `JsonDocument` before adding the `bootCount` member.
Something along those lines:

```c++
if (!doc.is<JsonObject>())
  doc.clear()
```

Now, should `JsonDocument` behave differently in step 8?
That's a good question.
The current implementation only transforms the document into an object if the document is empty.
I was satisfied with this behavior, and as far as I know, nobody had issues with it, but now I'm wondering if it's wise; maybe it should transform into an object even if the document is not empty.
I'll think about that.

Best regards,
Benoit

jrhodnik, 2022-10-15:
Goodness, what an interesting root cause. I do understand though. I don't see many people running into this so I'm not sure much action should be taken. I did eventually find that calling `clear` was the solution as well, however I left the bug open just in case. 

I wish I could title this bug better so others could find it.

closed 2022-10-15 by jrhodnik