bblanchon/ArduinoJson #1816: Please give an example of reading JSON for PubSubClient and sending JSON from HttpClient using buffer

butaikis, 2022-10-22:


bblanchon, 2022-10-23:
Hi @butaikis,

Did you look at the following pages?

* [How to use ArduinoJson with PubSubClient?](https://arduinojson.org/v6/how-to/use-arduinojson-with-pubsubclient/)
* [How to use ArduinoJson with HTTPClient?](https://arduinojson.org/v6/how-to/use-arduinojson-with-httpclient/)

butaikis, 2022-10-23:
> Hi @butaikis,
> 
> Did you look at the following pages?
> 
> * [How to use ArduinoJson with PubSubClient?](https://arduinojson.org/v6/how-to/use-arduinojson-with-pubsubclient/)
> * [How to use ArduinoJson with HTTPClient?](https://arduinojson.org/v6/how-to/use-arduinojson-with-httpclient/)
> 
> Best regards, Benoit

Hello. Yes of course.
PubSubClient, there is an example of how to send big data with the StreamUtils library. 
But I didn't find an example of how to read large messages.

HttpClient. There is an example of how to read the request response immediately in `deserializeJson(doc, httpClient.getStream());`. 
But how to send big data using StreamUtils ?

bblanchon, 2022-10-25:
Hi @butaikis,

To receive a large message with `PubSubClient`, you'll need to enlarge the internal buffer (it's only 256 bytes by default).
If the payload is composed of multiple messages, you can accumulate the data in a [`StringStream`](https://github.com/bblanchon/ArduinoStreamUtils#how-to-use-a-string-as-a-stream).

`HttpClient` doesn't let us use the stream interface for sending.
I recommend you use the `WiFiClient` instance directly, as shown in the [serialization tutorial](https://arduinojson.org/v6/doc/serialization/).
You can speed up the process with a `WriteBufferingStream`, as shown in [How to improve (de)serialization speed?](https://arduinojson.org/v6/how-to/improve-speed/#speed-up-serialization)

Best regards,
Benoit

butaikis, 2022-10-31:
Hello @bblanchon, thanks  for the reply.
Is it possible for an [ESPAsyncWebServer ](https://github.com/me-no-dev/ESPAsyncWebServer) to read a JSON request into a buffer? It uses an internal json buffer, and it is very small for big data.

bblanchon, 2022-11-03:
Yes.  You can accumulate the data in a [`StringStream`](https://github.com/bblanchon/ArduinoStreamUtils#how-to-use-a-string-as-a-stream) and call `deserializeJson()` when the message is complete.

butaikis, 2022-11-04:
> Yes. You can accumulate the data in a [`StringStream`](https://github.com/bblanchon/ArduinoStreamUtils#how-to-use-a-string-as-a-stream) and call `deserializeJson()` when the message is complete. See also #1813.

Thank you  @bblanchon. I'll try to use it.
I still have a task to calculate the hash sum of JSON. I was reading a ~40 KB file and the amount was correct. But by increasing the file to ~90 KB and it already gives broken/incorrect data. But if the file is output to the console, then the data matches. Is it possible to get rid of this problem ?

```
 DynamicJsonDocument OUTPUT_JSON1 {200000};
  File file = SPIFFS.open("/200kb.tst.txt", "r");
DeserializationError error = deserializeJson(OUTPUT_JSON1, file);
  file.close();
const char* data = OUTPUT_JSON1.as<String>().c_str();
size_t ss=1000;

  for (unsigned int i = 0; i < size_str ; i = i + ss, data = data + ss)
  {
    
     if (size_str - i < ss)
   {
      ss = size_str - i;
      
    }
    char buf[ss];
    memcpy(buf,  data, ss );
   mbedtls_md_update(&ctx, (const unsigned char *) buf, ss);
  }
```

bblanchon, 2022-11-08:
The following line creates a dangling pointer:

```
const char* data = OUTPUT_JSON1.as<String>().c_str();
```

Don't call `c_str()` on a temporary string, or you'll have a [use-after-free](https://cwe.mitre.org/data/definitions/416.html) bug.
If you have no idea what I'm talking about, consider reading "The Missing C++ Course" in [my book](https://arduinojson.org/book/).

bblanchon, 2022-11-08:
See also: [How to compute the hash of a JSON document?](https://arduinojson.org/v6/how-to/compute-hash/)

butaikis, 2022-11-08:
@bblanchon thank you very much for the clarification, and for the example of hash calculation.
Please tell me in which cases it will be called

bblanchon, 2022-11-12:
It uses `write(const uint8_t *, size_t)` when it can and `write(uint8_t)` when it must.
The JSON serializer rarely uses the "buffer" version but the MessagePack serializer uses it a lot.

closed 2022-11-26 by bblanchon